realsenseai/realsense-ros #258: [zr300] frame_id prefixing not correct for ir, ir2, imu and fisheye

ipa-fxm, 2017-07-26:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 14.04.5               |
| Kernel                | ?-?-?                |
| ROS                   | ROS Indigo                    |
| ROS RealSense         | from source https://github.com/intel-ros/realsense/commit/9211801edeaecac1a7ebfa94c6517fd9d95fbb14              |
| librealsense          | current release                |
| Camera Type-Firmware  | zr300            |

---

### Expected Behavior
all frames broadcasted by realsense_camera driver are prefixed using the `camera` launch file argument

### Actual Behavior
frame prefixing works for `depth` and `rgbd` frames, but does not work for `ir`, `ir2`, `imu` and `fisheye`.
for those they seem to be (hardcoded/always) prefixed with "camera", i.e. the argument is not considered

topic prefixing works correctly for all image topics

### Steps to Reproduce
 - `roslaunch realsense_camera zr300_nodelet_rgdb.launch camer:=oink`

A `rostopic echo /oink/depth/camera_info/header -n1` (rgbd respectively) will show:
```
header:
  seq: 61
  stamp:
    secs: 1501081015
    nsecs: 824442857
  frame_id: oink_depth_optical_frame
```
while `rostopic echo /oink/fisheye/camera_info/header -n1` (ir, ir2, imu respectively) will show:
```
header:
  seq: 0
  stamp:
    secs: 1501081106
    nsecs: 903379461
  frame_id: camera_fisheye_optical_frame
```
the wrong prefixing can also be seen when useing `rosrun tf2_tools view_frames.py`!

ipa-fxm, 2017-09-06:
fix is proposed in #259

icarpis, 2017-09-12:
PR #259 is merged

closed 2017-09-12 by icarpis