realsenseai/realsense-ros #264: Unable to install RealSense ROS on Jetson TK1

indigomega, 2017-09-05:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | 14.04 (L4T 21.4)              |
| Kernel                | 3.10.40                |
| ROS                   | Indigo                    |
| ROS RealSense         | 1.8.0                |
| librealsense          | 1.12.1                |
| Camera Type-Firmware  | ?-?.?.?.?            |

### Expected Behavior

Smooth installation.

## Actual Behavior

After doing 'catkin_make -j1', all modules build ok, but when it starts 
Building CXX object gtest/CMakeFiles/gtest.dir/src/gtest-all.cc.o`, it then spews a lot of errors (too many for the terminal to keep track of actually, thus deleting some of the error logs), attached below:

`/usr/include/c++/4.8/bits/stl_bvector.h:293:3: note: std::_Bit_iterator std::operator+(std::ptrdiff_t, const std::_Bit_iterator&)
   operator+(ptrdiff_t __n, const _Bit_iterator& __x)
   ^
/usr/include/c++/4.8/bits/stl_bvector.h:293:3: note:   no known conversion for argument 1 from 'std::basic_string<char>' to 'std::ptrdiff_t {aka int}'
In file included from /usr/include/c++/4.8/bits/stl_algobase.h:67:0,
                 from /usr/include/c++/4.8/vector:60,
                 from /usr/include/gtest/gtest.h:55,
                 from /usr/src/gtest/src/gtest-all.cc:39:
/usr/include/c++/4.8/bits/stl_iterator.h:333:5: note: template<class _Iterator> std::reverse_iterator<_Iterator> std::operator+(typename std::reverse_iterator<_Iterator>::difference_type, const std::reverse_iterator<_Iterator>&)
     operator+(typename reverse_iterator<_Iterator>::difference_type __n,
     ^
/usr/include/c++/4.8/bits/stl_iterator.h:333:5: note:   template argument deduction/substitution failed:
In file included from /usr/src/gtest/src/gtest.cc:135:0,
                 from /usr/src/gtest/src/gtest-all.cc:42:
/usr/src/gtest/src/gtest-internal-inl.h:1186:67: note:   'testing::internal::String' is not derived from 'const std::reverse_iterator<_Iterator>'
            StreamableToString((test_info.result())->elapsed_time()) + "ms");
                                                                   ^
In file included from /usr/include/c++/4.8/bits/stl_algobase.h:67:0,
                 from /usr/include/c++/4.8/vector:60,
                 from /usr/include/gtest/gtest.h:55,
                 from /usr/src/gtest/src/gtest-all.cc:39:
/usr/include/c++/4.8/bits/stl_iterator.h:902:5: note: template<class _Iterator, class _Container> __gnu_cxx::__normal_iterator<_Iterator, _Container> __gnu_cxx::operator+(typename __gnu_cxx::__normal_iterator<_Iterator, _Container>::difference_type, const __gnu_cxx::__normal_iterator<_Iterator, _Container>&)
     operator+(typename __normal_iterator<_Iterator, _Container>::difference_type
     ^
/usr/include/c++/4.8/bits/stl_iterator.h:902:5: note:   template argument deduction/substitution failed:
In file included from /usr/src/gtest/src/gtest.cc:135:0,
                 from /usr/src/gtest/src/gtest-all.cc:42:
/usr/src/gtest/src/gtest-internal-inl.h:1186:67: note:   'testing::internal::String' is not derived from 'const __gnu_cxx::__normal_iterator<_Iterator, _Container>'
            StreamableToString((test_info.result())->elapsed_time()) + "ms");
                                                                   ^
/usr/src/gtest/src/gtest-internal-inl.h: In member function 'virtual void testing::internal::StreamingListener::OnTestPartResult(const testing::TestPartResult&)':
/usr/src/gtest/src/gtest-internal-inl.h:1194:21: error: no match for 'operator+' (operand types are 'std::basic_string<char>' and 'testing::internal::String')
            "&line=" + StreamableToString(test_part_result.line_number()) +
                     ^
/usr/src/gtest/src/gtest-internal-inl.h:1194:21: note: candidates are:
In file included from /usr/include/c++/4.8/string:52:0,
                 from /usr/include/c++/4.8/bits/locale_classes.h:40,
                 from /usr/include/c++/4.8/bits/ios_base.h:41,
                 from /usr/include/c++/4.8/ios:42,
                 from /usr/include/c++/4.8/ostream:38,
                 from /usr/include/c++/4.8/iostream:39,
                 from /usr/include/gtest/internal/gtest-port.h:196,
                 from /usr/include/gtest/internal/gtest-internal.h:40,
                 from /usr/include/gtest/gtest.h:57,
                 from /usr/src/gtest/src/gtest-all.cc:39:
/usr/include/c++/4.8/bits/basic_string.h:2418:5: note: template<class _CharT, class _Traits, class _Alloc> std::basic_string<_CharT, _Traits, _Alloc> std::operator+(const std::basic_string<_CharT, _Traits, _Alloc>&, _CharT)
     operator+(const basic_string<_CharT, _Traits, _Alloc>& __lhs, _CharT __rhs)
     ^
/usr/include/c++/4.8/bits/basic_string.h:2418:5: note:   template argument deduction/substitution failed:
[... 432 lines cut ...]
/usr/src/gtest/src/gtest.cc:2484:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
/usr/src/gtest/src/gtest.cc: In function 'std::string testing::internal::FormatEpochTimeInMillisAsIso8601(testing::internal::TimeInMillis)':
/usr/src/gtest/src/gtest.cc:3243:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
/usr/src/gtest/src/gtest.cc: At global scope:
/usr/src/gtest/src/gtest.cc:316:13: warning: 'bool testing::internal::GTestIsInitialized()' defined but not used [-Wunused-function]
 static bool GTestIsInitialized() { return g_init_gtest_count != 0; }
             ^
/usr/src/gtest/src/gtest.cc:2028:20: warning: 'std::string testing::internal::FormatCxxExceptionMessage(const char*, const char*)' defined but not used [-Wunused-function]
 static std::string FormatCxxExceptionMessage(const char* description,
                    ^
make[2]: *** [gtest/CMakeFiles/gtest.dir/src/gtest-all.cc.o] Error 1
make[1]: *** [gtest/CMakeFiles/gtest.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j1" failed
`

Is there any missing directory I should be mindful of? Please help, appreciate any advice.

### Steps to Reproduce

If you have a spare Jetson TK1 and an R200, then install librealsense as shown in the youtube video by JetsonHacks. Then, confirm it works fine (I confirmed this). Then, build the intel-ros RealSense package as shown in the instructions at http://wiki.ros.org/realsense_camera/Tutorials/Building_from_Sources. Keep in mind, my version is ROS Indigo and the board is flashed with 21.4.

icarpis, 2017-09-10:
Hi @indigomega, we haven't tried yet to build the Intel RealSense ROS package on Jetson.
According to the log I guess that GTest library is missing. please try to disable catkin unit tests by run `catkin_make -DCATKIN_ENABLE_TESTING=false`

indigomega, 2017-09-11:
Hey, I tried to add the gtest library but it constantly failed, so I disabled testing, but the building still wants to do testing as shown below:
`Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/realsense_camera/tests_camera_core
.
.
.
CMakeFiles/tests_camera_core.dir/test/camera_core.cpp.o:camera_core.cpp:(.text+0x1e3e): more undefined references to `testing::internal::AssertHelper::~AssertHelper()' follow
CMakeFiles/tests_camera_core.dir/test/camera_core.cpp.o: In function `RealsenseTests_testDepthStream_Test::TestBody()':
camera_core.cpp:(.text+0x1f3a): undefined reference to `testing::internal::GetBoolAssertionFailureMessage(testing::AssertionResult const&, char const*, char const*, char const*)'
camera_core.cpp:(.text+0x1f50): undefined reference to `testing::internal::AssertHelper::AssertHelper(testing::TestPartResult::Type, char const*, int, char const*)'
camera_core.cpp:(.text+0x1f60): undefined reference to `testing::internal::AssertHelper::operator=(testing::Message const&) const'
camera_core.cpp:(.text+0x1f6e): undefined reference to `testing::internal::AssertHelper::~AssertHelper()'
camera_core.cpp:(.text+0x1fa8): undefined reference to `testing::internal::GetBoolAssertionFailureMessage(testing::AssertionResult const&, char const*, char const*, char const*)'
camera_core.cpp:(.text+0x1fbe): undefined reference to `testing::internal::AssertHelper::AssertHelper(testing::TestPartResult::Type, char const*, int, char const*)'
camera_core.cpp:(.text+0x1fce): undefined reference to `testing::internal::AssertHelper::operator=(testing::Message const&) const'
camera_core.cpp:(.text+0x1fdc): undefined reference to `testing::internal::AssertHelper::~AssertHelper()'
camera_core.cpp:(.text+0x203e): undefined reference to `testing::internal::AssertHelper::AssertHelper(testing::TestPartResult::Type, char const*, int, char const*)'
camera_core.cpp:(.text+0x204c): undefined reference to `testing::internal::AssertHelper::operator=(testing::Message const&) const'
camera_core.cpp:(.text+0x205a): undefined reference to `testing::internal::AssertHelper::~AssertHelper()'
camera_core.cpp:(.text+0x2092): undefined reference to `testing::internal::GetBoolAssertionFailureMessage(testing::AssertionResult const&, char const*, char const*, char const*)'
camera_core.cpp:(.text+0x20a8): undefined reference to `testing::internal::AssertHelper::AssertHelper(testing::TestPartResult::Type, char const*, int, char const*)'
camera_core.cpp:(.text+0x20b8): undefined reference to `testing::internal::AssertHelper::operator=(testing::Message const&) const'
camera_core.cpp:(.text+0x20c6): undefined reference to `testing::internal::AssertHelper::~AssertHelper()'
camera_core.cpp:(.text+0x2114): undefined reference to `testing::internal::AssertHelper::AssertHelper(testing::TestPartResult::Type, char const*, int, char const*)'
camera_core.cpp:(.text+0x2122): undefined reference to `testing::internal::AssertHelper::operator=(testing::Message const&) const'
camera_core.cpp:(.text+0x2130): undefined reference to `testing::internal::AssertHelper::~AssertHelper()'
camera_core.cpp:(.text+0x21f0): undefined reference to `testing::internal::AssertHelper::~AssertHelper()'
camera_core.cpp:(.text+0x2204): undefined reference to `testing::internal::AssertHelper::~AssertHelper()'
camera_core.cpp:(.text+0x221e): undefined reference to `testing::internal::AssertHelper::~AssertHelper()'
camera_core.cpp:(.text+0x2264): undefined reference to `testing::internal::AssertHelper::~AssertHelper()'
CMakeFiles/tests_camera_core.dir/test/camera_core.cpp.o:camera_core.cpp:(.text+0x2286): more undefined references to `testing::internal::AssertHelper::~AssertHelper()' follow
CMakeFiles/tests_camera_core.dir/test/camera_core.cpp.o: In function `RealsenseTests_testInfrared1Stream_Test::TestBody()':
camera_core.cpp:(.text+0x23a0): undefined reference to `testing::internal::GetBoolAssertionFailureMessage(testing::AssertionResult const&, char const*, char const*, char const*)'
camera_core.cpp:(.text+0x23b6): undefined reference to `testing::internal::AssertHelper::AssertHelper(testing::TestPartResult::Type, char const*, int, char const*)'
camera_core.cpp:(.text+0x23c6): undefined reference to `testing::internal::AssertHelper::operator=(testing::Message const&) const'
camera_core.cpp:(.text+0x23d4): undefined reference to `testing::internal::AssertHelper::~AssertHelper()'
camera_core.cpp:(.text+0x240e): undefined reference to `testing::internal::GetBoolAssertionFailureMessage(testing::AssertionResult const&, char const*, char const*, char const*)'
camera_core.cpp:(.text+0x2424): undefined reference to `testing::internal::AssertHelper::AssertHelper(testing::TestPartResult::Type, char const*, int, char const*)'
camera_core.cpp:(.text+0x2434): undefined reference to `testing::internal::AssertHelper::operator=(testing::Message const&) const'
camera_core.cpp:(.text+0x2442): undefined reference to `testing::internal::AssertHelper::~AssertHelper()'
camera_core.cpp:(.text+0x249a): undefined reference to `testing::internal::GetBoolAssertionFailureMessage(testing::AssertionResult const&, char const*, char const*, char const*)'
camera_core.cpp:(.text+0x24b0): undefined reference to `testing::internal::AssertHelper::AssertHelper(testing::TestPartResult::Type, char const*, int, char const*)'
camera_core.cpp:(.text+0x24c0): undefined reference to `testing::internal::AssertHelper::operator=(testing::Message const&) const'
[... cut ...]
CMakeFiles/tests_camera_core.dir/test/camera_core.cpp.o:(.data.rel.ro+0x2b0): undefined reference to `testing::Test::SetUp()'
CMakeFiles/tests_camera_core.dir/test/camera_core.cpp.o:(.data.rel.ro+0x2b4): undefined reference to `testing::Test::TearDown()'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/ubuntu/catkin_ws/devel/lib/realsense_camera/tests_camera_core] Error 1
make[1]: *** [realsense/realsense_camera/CMakeFiles/tests_camera_core.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j1" failed`

Any ideas?

icarpis, 2017-09-13:
I managed to install RealSense ROS from sources on Jetson TX 2 with Ubuntu 16.04 and Kernel 4.4.38-tegra.
At first, I got the same gtest linkage errors when I tried to build it only with catkin_make command, but when I disabled unit tests (catkin_make -DCATKIN_ENABLE_TESTING=false) it compiled without any errors.

Maybe cleaning the catkin workspace will solve the issue
> catkin_make clean
> catkin_make -DCATKIN_ENABLE_TESTING=false

indigomega, 2017-09-14:
Hey icarpis, thank you for testing it out with your machine. Unfortunately, I kind of gave up on working with the TK1. I've had too much difficulty with it and I guess its prime is gone, since the support is virtually nonexistent and its kernel version is way outdated. Ordered an UP board, will see how that goes!

closed 2017-09-14 by indigomega