realsenseai/realsense-ros #265: Support for ROS2

ohshout, 2017-09-06:
Hi friends,

I wonder if there's any plan of having ROS2 realsense camera packages? I'm specifically looking for using the R200 camera on Ubuntu with ROS2 on the Intel Aero board.

And how hard you think it would be to port the realsense ROS1 packages to ROS2? Any obvious obstacle? Want to hear some thoughts from you experts :)

Thanks a lot!
Tom.

ohshout, 2017-09-06:
I found this guide to migrate from ROS1 to ROS2: https://github.com/ros2/ros2/wiki/Migration-Guide
Do you think this applies to realsense package as well?

testkit, 2018-02-07:
Realsense ROS2 migration is in progress, we target to get it ready in 3 weeks or so.

zenving, 2018-02-21:
@testkit  Hi, have you done that ?

testkit, 2018-03-26:
Finally, it's published at https://github.com/intel/ros2_intel_realsense.

closed 2018-03-26 by testkit