realsenseai/realsense-ros #267: Trigger new release for Indigo and Kinetic

ipa-fxm, 2017-09-22:
Dear package maintainers,

since the last release (1.8.0) several features and fixes (in particular #244 and #259) have been added to this repo that we would need to release our own repos.
Do you intend to trigger a new release - for both Indigo and Kinetic - some time soon?
Thanks!

ipa-fxm, 2017-10-20:
Hi guys,
any intention on triggering a new release for this repo?
is there anything blocking a new release atm?
Thanks

ipa-fxm, 2017-12-12:
@icarpis @dorodnic 
we are still in need of a new release for indigo and kinetic due to the launch file renaming - which is blocking our release of `cob_robots`...do you see any chance in releasing this update before christmas?

icarpis, 2017-12-13:
Hello @ipa-fxm, a new version of RealSense ROS 1 for indigo and kinetic will be released soon but I can't guarantee that it will be before Christmas.

icarpis, 2017-12-27:
New version of RealSense ROS 1 for indigo and kinetic is [available](http://repositories.ros.org/status_page/ros_kinetic_default.html?q=realsense)

closed 2017-12-27 by icarpis