realsenseai/realsense-ros #268: registering realsense data during playback

mauricefallon, 2017-09-27:
### System Configuration

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      |  Ubuntu 16.04.3 LTS                |
| Kernel                | 4.10.0-35-generic              |
| ROS                   | kinetic                    |
| ROS RealSense         | 1.8.0                |
| librealsense          | 1.12.1                |
| Camera Type-Firmware  | Intel RealSense SR300 - 3.10.10.0 |

---

This is a general question. [Its mentioned here that creating registered point clouds live from the device is very expensive](http://wiki.ros.org/realsense_camera#Launching_Multiple_Cameras) under section 4.4. It's uses about 4 times more bandwidth (167MB/s)

Instead, I want to collect ROSbags with image_raw for both color and depth and compute the registration offline. Are there any demos or examples for doing this?

skohlbr, 2017-09-28:
Haven't used the SR300 (only R200), but essentially you'd have to record the data you mention, play it back and start the [processing.launch.xml](https://github.com/ros-drivers/rgbd_launch/blob/indigo-devel/launch/includes/processing.launch.xml) launch file from the rgbd_launch package. This is otherwise started along with the standard driver setup here in [line 91 of sr300_nodelet_rgbd.launch](https://github.com/intel-ros/realsense/blob/indigo-devel/realsense_camera/launch/sr300_nodelet_rgbd.launch#L91). This approach generates the point cloud using ROS nodelets outside the Realsense driver, which is the default since #47 from my understanding.

icarpis, 2017-10-18:
Hello @mauricefallon, there isn't such of example, but you can record the requested topics to a ROS bag file using [rqt_bag](http://wiki.ros.org/rqt_bag) tool. First, start the sr300_nodelet_default.launch and then open rqt_bag to record the topics.

closed 2017-10-31 by icarpis