realsenseai/realsense-ros #272: Aligned RGB and depth image for librealsense 2.0

lz89, 2017-10-30:
### System Configuration
| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 16.04.3  |
| Kernel                | 4.10.0-37 |
| ROS                   | kinetic |
| ROS RealSense         | v2.0.0 |
| librealsense          | v2.8.0 |
| Camera Type-Firmware  |SR300-1.17|

---
I am using librealsense-2 and realsense_ros_camera with tag 2.0.
How can I get aligned colour and depth image, as well as RGB cloud?
I understand it's not supported by the current version yet.
But would you please give me some pointers so that I can implement myself?

icarpis, 2017-10-31:
Hello @lz89, the RGB point-cloud will be provided at the next ROS wrapper version in a couple of days.
If you still want to implement it by yourself please take a look at [this ](https://github.com/intel-ros/realsense/blob/9211801edeaecac1a7ebfa94c6517fd9d95fbb14/realsense_camera/src/base_nodelet.cpp#L925) function for a reference. Pay attention that it's above the [old LibRealSense](https://github.com/IntelRealSense/librealsense/tree/legacy) API which means that it won't work out of the box above [LibRealSense 2](https://github.com/IntelRealSense/librealsense/tree/master) API.

lz89, 2017-10-31:
@icarpis In that case I will wait for the new release.

closed 2017-10-31 by lz89