realsenseai/realsense-ros #277: realsense 2 aligned depth and point cloud based on color image

lz89, 2017-11-17:
### System Configuration
| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 16.04.3  |
| Kernel                | 4.10.0-37 |
| ROS                   | kinetic |
| ROS RealSense         | v2.0.1 |
| librealsense          | v2.8.0 |
| Camera Type-Firmware  |SR300-3.17.0.0|

---
I am using librealsense-2 and realsense_ros_camera development branch (with RGB point cloud).
I wonder if there is a way to enable the aligned depth to be published to ROS.
Currently, the RGB point cloud (/camera/points) is an organised point cloud stored (arranged) in the IR image order. This is because in 'realsense_camera_node.cpp', for each pixel in the IR image, it reproject the pixel to 3D and transform to color camera coordinate then find the color by projecting the points on color image.
I would like to have: an organised point cloud arranged in the color image. i.e. for each pixel in the color image, I know its depth and the corresponding 3D point.

icarpis, 2017-11-21:
Hello @lz89, we will publish aligned_depth_to_color topic in the next RealSense ROS versions.

akashjinandra, 2018-01-17:
Is there an update to this?

icarpis, 2018-01-18:
This feature is still in-progress on my [repository](https://github.com/icarpis/realsense/commits/development).
I would be glad to get your feedback on it.
In order to get the aligned topic please run the `frames_aligned_to_depth` launch file.

icarpis, 2018-01-25:
I merged the alignment feature to the [development branch](https://github.com/intel-ros/realsense/tree/development).
See [here](https://github.com/intel-ros/realsense/tree/development#aligned-depth-frames) how to enable this feature.

icarpis, 2018-01-31:
Provided by [ROS wrapper 2.0.2](https://github.com/intel-ros/realsense/tree/2.0.2)

closed 2018-01-31 by icarpis

lz89, 2018-02-02:
@icarpis The new version (2.0.2) only provide the depth aligned to color but the point cloud is still in depth camera coordinate. I would like to have the point cloud in the color camera coordinate. That means for each pixel in the color image, I have the corresponding 3D position (in camera coordinate).

icarpis, 2018-02-05:
@lz89, the point cloud provided by the [rs_rgbd](https://github.com/intel-ros/realsense/tree/2.0.2#rgbd-point-cloud) launch file is created by the `aligned-depth-to-color` topic and by the `color` topic. 
The `aligned-depth-to-color` topic is match exactly to the color topic, so they have the same coordinate system, and therefore the point cloud topic is in **color coordinate** system.

In order to retrieve the 3D position of each color pixel I would suggest you to go over the `aligned-depth-to-color` pixels and for each pixel that is different than zero, taking its corresponding pixel in the `color` image. see below code snippet

```cpp
const int size_of_depth_pixel = 2; // pixel size in bytes
const int size_of_color_pixel = 3; // pixel size in bytes
uint8_t* aligned_depth_to_color;
uint8_t* color;
for (int y = 0; y < depth_frame_height; ++y)
{
	for (int x = 0; x < depth_frame_width; ++x)
	{
		int depth_pixel_position = (y * depth_frame_width + x) * size_of_depth_pixel;
		if (aligned_depth_to_color[depth_pixel_position] != 0)
		{
			int color_pixel_position = (y * depth_frame_width + x) * size_of_color_pixel;
			corresponding_color_pixel = color[color_pixel_position];
		}
	}
}
```

lz89, 2018-02-13:
@icarpis Thank you for the clarification of the rs_rgbd launch file.
However, when I subscribe to the topic `/camera/depth_registered/points`, I found that the x, y, z value of the point in the cloud is strange. It looks neither in meter nor millimeter. I wonder if the cloud published by this topic has considered the depth scale.

Following is how I get the point cloud in the ROS callback:
```
void Listener::pointCloudCallback(const sensor_msgs::PointCloud2ConstPtr &msg) {
    std::lock_guard<std::mutex> lock(m_mx_cloud);
    pcl::fromROSMsg(*msg, *m_cloud);
}
```

icarpis, 2018-02-15:
The point cloud is published by [depth_image_proc/point_cloud_xyzrgb](http://wiki.ros.org/depth_image_proc#depth_image_proc.2BAC8-point_cloud_xyzrgb) nodelet.
Acording to the RGBD nodelet [implementation](https://github.com/ros-perception/image_pipeline/blob/2c8a054ad1aff0acdac9af058812cf106e41a8e9/depth_image_proc/src/nodelets/point_cloud_xyzrgb.cpp#L329), it looks that the units of depth values (z) are converted from mm into meters.
Could you please elaborate how do you measure the distance with RViz ?

lz89, 2018-02-15:
@icarpis  I put a plane in the view and measure the distance from the camera to the plane (plane is put almost parallel to the camera X-Y plane). Then I added an axis (coordinate) in RViz which I can adjust the length of the axis so that the Z axis just intersect with the plane.

icarpis, 2018-02-22:
@lz89, the `depth_scale` of D400 devices is 0.001.
The `RGBD` assume that the depth data haven't been multiplied by the `depth_scale` and therefore convert the values from millimeters to meters by [multiplying](https://github.com/ros-perception/image_pipeline/blob/2589ed28251268a9657f4cc037373b063cca9958/depth_image_proc/include/depth_image_proc/depth_traits.h#L49) the raw data by 0.001.

lz89, 2018-02-22:
@icarpis Thanks for the reply. So in case of SR300, what shall I do to correct this?

icarpis, 2018-02-22:
In case of SR300, I think that you need to multiply the depth data by 0.1 because the `depth scale` is 0.0001.

lz89, 2018-02-22:
@icarpis I once read it from the librealsense API, which gave me a value like 0.0025 (or 0.00025). If the RGBD nodelet assume the scaling is 0.001 (while the actual one is not), wouldn't it make the X and Y value being scaled as well?

icarpis, 2018-02-22:
No, the depth_scale doesn't effect on the X and Y axes values but only on the Z axis values.

YeeCY, 2019-08-27:
@lz89 Any suggestion on how to modify the `depth scale` in realsense-ros?

doronhi, 2019-08-28:
You can't change the depth scale in the realsense-ros. According to the [ROS standard](https://www.ros.org/reps/rep-0118.html#id4), the depth is in mm.
Suppose we allowed the modification of the depth scale. We would have ended up scaling it back to mm.
As a side note, I think that given the mechanical issues, i.e. the resolution of the sensors, the distance between them and the scale of the problem in general, there is no gain by changing the scale of the the depth.

YeeCY, 2019-08-28:
> You can't change the depth scale in the realsense-ros. According to the [ROS standard](https://www.ros.org/reps/rep-0118.html#id4), the depth is in mm.
> Suppose we allowed the modification of the [...]

Oh, thanks for the notification! Actually, I'm working with a D435 dynamically mounted on the wrist of a UR5. It turns out launching `rgbd.launch` gives me a aligned pointcloud. I also find that there are some `NAN` in the aligned pointcloud which I have to remove manually.