realsenseai/realsense-ros #280: Build issues with 2.0.1

thepaintedsnipe, 2017-12-03:
### System Configuration

OS:  Ubuntu 16.04.5
kernel 4.10.0-40-generic
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.5) 5.4.0 20160609
ROS Kinetic 
librealsense:  v2.8.1 & v2.8.2
ROS realsense wrapper: 2.0.0 & 2.0.1
Camera is D430 with firmware 5.8.9.0 but problem is with build

### Expected Behavior
Realsense 2.0.1 should build fine

### Actual Behavior
Realsense 2.0.1 build fails with following error

```CMake Warning at realsense-2.0.1/realsense_ros_camera/CMakeLists.txt:15 (find_package):
  By not providing "Findrealsense2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "realsense2", but CMake did not find one.

  Could not find a package configuration file provided by "realsense2" with
  any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.

CMake Error at realsense-2.0.1/realsense_ros_camera/CMakeLists.txt:17 (message):
  
   Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases
``` 

### Steps to Reproduce

icarpis, 2017-12-03:
Hello @thepaintedsnipe, it looks that you tried to build the Intel ROS wrapper before installing the [RealSense SDK 2.0](https://github.com/IntelRealSense/librealsense/releases).
Please follow the [installation instructions](https://github.com/intel-ros/realsense/tree/2.0.1#step-1-install-the-latest-intel-realsense-sdk-20) of Intel ROS wrapper.

thepaintedsnipe, 2017-12-03:
Hi Icarpis,

   Thanks for your suggestion. Sorry that I didn't elaborate on that part. I do have RealSense SDK installed. Before trying to install realsense, I confirmed that the librealsense is installed.
```
catkin_ws-> rospack list | grep real
librealsense2 /home/kunal/kunal_workspace/aivero/ROS_WS/catkin_ws/src/librealsense-2.8.1
``` 
I had in past installed *librealsense-ros1.12.1* and *reallsense-1*  which didn't show any problems. So I uninstalled the librealsense2 and realsense2 and tried the above combination again. Build went smooth. This given me an impression that my build set up and build procedure is correct. Possibly it may be due to having a wrong version or wrong configuration?

thepaintedsnipe, 2017-12-03:
Hello icarpis,
I again tried to install the 2 libraries as per the instruction which you shared
```(1) First removed both librealsense2 and realsense2 from my catkin_ws
(2) Then installed RealSense SDK 2.0 outside the catkin_ws using the installation instructions.
(3) Then added realsense2 to catkin_ws and tried build.
``` 
It went fine. Thanks for your help. 
I was tricked here because for old librealsense-1 and realsense-1.0 combination, installing from catkin_ws went fine. 
Thanks for your help

closed 2017-12-03 by thepaintedsnipe

AndyZe, 2018-03-16:
I got the same error when trying to install from Debian, following these instructions:
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

ROS kinetic
kernel 4.13.0-37-generic

ConnorChristie, 2019-09-08:
You can also try deleting the `build` folder in your catkin_ws then re-running `catkin_make`. This worked for me