realsenseai/realsense-ros #281: visual parameter preset defined in realsense-viewer but not in realsense-2.0.1 package

poripsa, 2017-12-12:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      |Ubuntu 16.04.3 LTS |
| Kernel                | 4.4.0-101-generic |
| ROS                   |kinetic |
| ROS realsense_ros_camera         | 2.0.1|
| ROS RealSense         | 1.8.0|
| LibRealSense         | 2.8.1| >>used by realsense_ros_camera
| librealsense          | 1.12.1|
| Camera Type-Firmware  | 05.08.14.00 |

poripsa, 2017-12-12:
[image]

poripsa, 2017-12-12:
How can the visual preset parameter be modified in the ros launch file (or anywhere else) so that it loads when camera is launched

icarpis, 2017-12-13:
Hello @poripsa, currently visual presets cannot be modified by RealSense ROS wrapper or Launch file. we will implement it in the next release.
For a workaround you can change the visual preset using 'realsense-viewer' tool and then use the ROS wrapper/Launch file. Please note that in order to set a new preset you should streaming with Depth.
Modify a new preset will remains as long as the camera stay connected to the computer.

poripsa, 2017-12-15:
Will the same preset be applied when the camera is reconnected or after start up of computer? How can i save the settings and pull them up as required?

icarpis, 2017-12-16:
No, when the camera is reconnected the default preset is loaded. You can use the 'realsense-viewer' tool to save and load camera settings using JSON file.
[image]

The JSON file contains the applied preset and camera controls.

poripsa, 2018-01-02:
Hey I am looking to set this up for my project. Any pointers you could give me? So far I see the rosnode just publishes the set (by preset) frames.

shaun-edwards, 2018-01-09:
Addressed in #289.  This issue can be closed now.

closed 2018-01-10 by icarpis

stevensu1838, 2019-08-16:
Hi all,

Can the visual preset parameter be modified in the ros launch file (or anywhere else) so that it loads when camera is launched yet or not? 
I am having the linked [issue](https://github.com/IntelRealSense/realsense-ros/issues/892)