realsenseai/realsense-ros #282: IMU coordinate frame in ZR300

tuandle, 2017-12-14:
| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 16.04.3 LTS                |
| Kernel                | 4.10.0-42-generic               |
| ROS                   | Kinetic                   |
| ROS RealSense         | 1.8.0               |
| librealsense          | 1.12.1                |
| Camera Type-Firmware  | 2.0.71.8            |

Hi,
I have a question regarding the coordinate's frame of the IMU in ZR300. 
I couldn't find any reference to this coordinate in the device's documentation. I wonder if there were any information available?
Thank you for your help!

icarpis, 2017-12-18:
The following image describes the coordinates of ZR300 Gyroscope and Accelerometer sensors.
[image]

Accelerometer axes: the positive x-axis points to the right, the positive y-axis points down and the positive z-axis points forward.
The gyrosocpe axes is around the accelerometer axes respectively.

tuandle, 2017-12-18:
Hi @icarpis ,
Thank you for your clarification!
Best,
Tuan

closed 2017-12-18 by tuandle