realsenseai/realsense-ros #283: noise

derekhe89, 2017-12-15:
Hello,i want to know noise parameter which geted form sdk is contious or  dispersed？

icarpis, 2017-12-17:
Hello @derekhe89, what do you mean by 'noise parameter' ?

derekhe89, 2017-12-20:
The noise parameter is the noise_variances and the bias_variances.

derekhe89, 2017-12-20:
I want konw the contious noise_density and random_walk of gyroscope and accelerometer,but i don't know how to get that . Is the noise_variances is noise_density and the bias_variances is random_walk?@ icarpis

icarpis, 2017-12-20:
The ROS wrapper publish the IMU calibration info in order to calibrate the IMU (Accel/Gyro). Please take a look at this issue #256 .

derekhe89, 2017-12-21:
Thank you for your answer. The issue you gave me just explain the scale、 cross axis and bias, but i want to know the contious noise_density and random_walk of gyroscope and accelerometer. @icarpis

icarpis, 2017-12-25:
These parameters could be found in this [issue](https://github.com/IntelRealSense/librealsense/issues/798).
If you still have more questions about the ZR300 IMU sensor please consider to post a question in that issue.

closed 2017-12-25 by icarpis