realsenseai/realsense-ros #284: compile error

jixiedaima, 2017-12-17:
/usr/bin/ld: /usr/local/lib/libgtest.a(gtest-all.cc.o): relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libgtest.a: 无法添加符号: 错误的值
/usr/bin/ld: /usr/local/lib/libgtest.a(gtest-all.cc.o): relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libgtest.a: 无法添加符号: 错误的值
collect2: error: ld returned 1 exit status
collect2: error: ld returned 1 exit status
realsense/realsense_camera/CMakeFiles/tests_rgbd_topics.dir/build.make:324: recipe for target '/home/y700/catkin_ws/devel/lib/realsense_camera/tests_rgbd_topics' failed
realsense/realsense_camera/CMakeFiles/tests_camera_core.dir/build.make:324: recipe for target '/home/y700/catkin_ws/devel/lib/realsense_camera/tests_camera_core' failed
make[2]: *** [/home/y700/catkin_ws/devel/lib/realsense_camera/tests_rgbd_topics] Error 1
make[2]: *** [/home/y700/catkin_ws/devel/lib/realsense_camera/tests_camera_core] Error 1
CMakeFiles/Makefile2:434: recipe for target 'realsense/realsense_camera/CMakeFiles/tests_rgbd_topics.dir/all' failed
make[1]: *** [realsense/realsense_camera/CMakeFiles/tests_rgbd_topics.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
CMakeFiles/Makefile2:3344: recipe for target 'realsense/realsense_camera/CMakeFiles/tests_camera_core.dir/all' failed
make[1]: *** [realsense/realsense_camera/CMakeFiles/tests_camera_core.dir/all] Error 2
[ 98%] Built target get_debug_info
[100%] Linking CXX shared library /home/y700/catkin_ws/devel/lib/librealsense_camera_nodelet.so
[100%] Built target realsense_camera_nodelet
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

My platform is ubuntu16.04!

icarpis, 2017-12-17:
Hello @jixiedaima, it seems that Catkin couldn't find the GTest library.
Please try to build without tests. `catkin_make -DCATKIN_ENABLE_TESTING=false`

jixiedaima, 2017-12-17:
Thank you so much, I solved it !

closed 2017-12-20 by icarpis