realsenseai/realsense-ros #288: can't launch the file (ZR300  )

jixiedaima, 2017-12-26:
$roslaunch realsense_camera zr300_nodelet_rgbd.launch
Intel RealSense F200_camera ; 2.60.0.0
Intel RealSense LR200_camera ; 2.0.71.18
Intel RealSense R200_camera ; 1.0.72.06
Intel RealSense SR300_camera ; 3.10.10.0
Intel RealSense ZR300_adapter ; 1.29.0.0
Intel RealSense ZR300_camera ; 2.0.71.28
Intel RealSense ZR300_motion_module ; 1.25.0.0
[ERROR] [1514259318.699326742]: /camera/driver - Error calling rs_get_device_info ( device:0x20e6360, info:ADAPTER_BOARD_FIRMWARE_VERSION ): 
selected camera info is not supported for this camera! 

[FATAL] [1514259318.838373675]: Failed to load nodelet '/camera/driver` of type `realsense_camera/ZR300Nodelet` to manager `camera_nodelet_manager'
[FATAL] [1514259318.838373488]: Failed to load nodelet '/camera/ir_rectify_ir` of type `image_proc/rectify` to manager `camera_nodelet_manager'
[FATAL] [1514259318.838385464]: Failed to load nodelet '/camera/depth_metric_rect` of type `depth_image_proc/convert_metric` to manager `camera_nodelet_manager'
[FATAL] [1514259318.838459011]: Failed to load nodelet '/camera/depth_metric` of type `depth_image_proc/convert_metric` to manager `camera_nodelet_manager'
[FATAL] [1514259318.838668944]: Failed to load nodelet '/camera/rgb_rectify_color` of type `image_proc/rectify` to manager `camera_nodelet_manager'
[FATAL] [1514259318.838770345]: Failed to load nodelet '/camera/depth_points` of type `depth_image_proc/point_cloud_xyz` to manager `camera_nodelet_manager'
[FATAL] [1514259318.838969271]: Failed to load nodelet '/camera/points_xyzrgb_sw_registered` of type `depth_image_proc/point_cloud_xyzrgb` to manager `camera_nodelet_manager'
[FATAL] [1514259318.839022437]: Failed to load nodelet '/camera/depth_rectify_depth` of type `image_proc/rectify` to manager `camera_nodelet_manager'
[FATAL] [1514259318.839178270]: Failed to load nodelet '/camera/register_depth_rgb` of type `depth_image_proc/register` to manager `camera_nodelet_manager'
[FATAL] [1514259318.839676947]: Failed to load nodelet '/camera/disparity_depth` of type `depth_image_proc/disparity` to manager `camera_nodelet_manager'
[FATAL] [1514259318.839717413]: Failed to load nodelet '/camera/disparity_registered_sw` of type `depth_image_proc/disparity` to manager `camera_nodelet_manager'
[FATAL] [1514259318.839942695]: Failed to load nodelet '/camera/depth_registered_sw_metric_rect` of type `depth_image_proc/convert_metric` to manager `camera_nodelet_manager'
[FATAL] [1514259318.844770843]: Failed to load nodelet '/camera/rgb_debayer` of type `image_proc/debayer` to manager `camera_nodelet_manager'
[camera/camera_nodelet_manager-1] process has died [pid 12402, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=camera_nodelet_manager __log:=/home/y700/.ros/log/3a601d0c-e9e6-11e7-8f29-e4a7a0332039/camera-camera_nodelet_manager-1.log].

icarpis, 2017-12-26:
Hello @jixiedaima, could you please do the following and provide the output?
1. Plug the camera into the computer
1. Open a Terminal
2. Type `cpp-enumerate-devices`

jixiedaima, 2017-12-26:
rs.warn: Motion module capabilities were disabled due to failure to acquire intrinsic
Device 0 - Intel RealSense ZR300:
 Serial number: 3511803652
 Firmware version: 2.0.71.28
 USB Port ID: 2-1-1
 Camera info: 
    DEVICE_NAME         : 	Intel RealSense ZR300
    DEVICE_SERIAL_NUMBER: 	3511803652
    CAMERA_FIRMWARE_VERSION: 	2.0.71.28
    CAMERA_TYPE         : 	PRQ-Ready
    OEM_ID              : 	OEM None
    ISP_FW_VERSION      : 	0xd942
    CONTENT_VERSION     : 	12
    MODULE_VERSION      : 	4.4.0.111
    IMAGER_MODEL_NUMBER : 	33
    CALIBRATION_DATE    : 	2014-07-04 08:18:35 UTC
    EMITTER_TYPE        : 	Laser Driver 4.1
    FOCUS_VALUE         : 	0
    LENS_TYPE           : 	Newmax 58.9 x 45.9 degs in VGA
    3RD_LENS_TYPE       : 	AEOT
    LENS_COATING__TYPE  : 	Visible-light block / IR pass 43 nm width
    3RD_LENS_COATING_TYPE: 	IR coating
    NOMINAL_BASELINE    : 	70 mm
    3RD_NOMINAL_BASELINE: 	58 mm
 Supported options:                                    min        max       step  default  
    COLOR_BACKLIGHT_COMPENSATION                       : 0    ... 1           1     0         
    COLOR_BRIGHTNESS                                   : -64  ... 64          1     0         
    COLOR_CONTRAST                                     : 0    ... 100         1     50        
    COLOR_EXPOSURE                                     : 50   ... 666         1     166       
    COLOR_GAIN                                         : 0    ... 128         1     64        
    COLOR_GAMMA                                        : 100  ... 500         1     300       
    COLOR_HUE                                          : -180 ... 180         1     0         
    COLOR_SATURATION                                   : 0    ... 100         1     64        
    COLOR_SHARPNESS                                    : 0    ... 100         1     50        
    COLOR_WHITE_BALANCE                                : 2800 ... 6500        10    4600      
    COLOR_ENABLE_AUTO_EXPOSURE                         : 0    ... 1           1     1         
    COLOR_ENABLE_AUTO_WHITE_BALANCE                    : 0    ... 1           1     1         
    R200_LR_AUTO_EXPOSURE_ENABLED                      : 0    ... 1           1     0         
    R200_LR_GAIN                                       : 100  ... 6399        1     400       
    R200_LR_EXPOSURE                                   : 1    ... 330         1     164       
    R200_EMITTER_ENABLED                               : 0    ... 1           1     0         
    R200_DEPTH_UNITS                                   : 0    ... 2.14748e+09 1     1000      
    R200_DEPTH_CLAMP_MIN                               : 0    ... 65535       1     0         
    R200_DEPTH_CLAMP_MAX                               : 0    ... 65535       1     65535     
    R200_DEPTH_CONTROL_ESTIMATE_MEDIAN_DECREMENT       : 0    ... 255         1     5         
    R200_DEPTH_CONTROL_ESTIMATE_MEDIAN_INCREMENT       : 0    ... 255         1     5         
    R200_DEPTH_CONTROL_MEDIAN_THRESHOLD                : 0    ... 1023        1     192       
    R200_DEPTH_CONTROL_SCORE_MINIMUM_THRESHOLD         : 0    ... 1023        1     1         
    R200_DEPTH_CONTROL_SCORE_MAXIMUM_THRESHOLD         : 0    ... 1023        1     512       
    R200_DEPTH_CONTROL_TEXTURE_COUNT_THRESHOLD         : 0    ... 31          1     6         
    R200_DEPTH_CONTROL_TEXTURE_DIFFERENCE_THRESHOLD    : 0    ... 1023        1     24        
    R200_DEPTH_CONTROL_SECOND_PEAK_THRESHOLD           : 0    ... 1023        1     27        
    R200_DEPTH_CONTROL_NEIGHBOR_THRESHOLD              : 0    ... 1023        1     7         
    R200_DEPTH_CONTROL_LR_THRESHOLD                    : 0    ... 2047        1     24        
    FRAMES_QUEUE_SIZE                                  : 1    ... 20          1     20        
 Stream DEPTH - 36 modes:
  640	x 480	@ 90Hz	Z16	57.7 x 44.9 degrees, distortion = NONE
  640	x 480	@ 90Hz	DISPARITY16	57.7 x 44.9 degrees, distortion = NONE
  640	x 480	@ 60Hz	Z16	57.7 x 44.9 degrees, distortion = NONE
  640	x 480	@ 60Hz	DISPARITY16	57.7 x 44.9 degrees, distortion = NONE
  640	x 480	@ 30Hz	Z16	57.7 x 44.9 degrees, distortion = NONE
  640	x 480	@ 30Hz	DISPARITY16	57.7 x 44.9 degrees, distortion = NONE
  628	x 468	@ 90Hz	Z16	56.8 x 43.9 degrees, distortion = NONE
  628	x 468	@ 90Hz	DISPARITY16	56.8 x 43.9 degrees, distortion = NONE
  628	x 468	@ 60Hz	Z16	56.8 x 43.9 degrees, distortion = NONE
  628	x 468	@ 60Hz	DISPARITY16	56.8 x 43.9 degrees, distortion = NONE
  628	x 468	@ 30Hz	Z16	56.8 x 43.9 degrees, distortion = NONE
  628	x 468	@ 30Hz	DISPARITY16	56.8 x 43.9 degrees, distortion = NONE
  492	x 372	@ 90Hz	Z16	57.4 x 45 degrees, distortion = NONE
  492	x 372	@ 90Hz	DISPARITY16	57.4 x 45 degrees, distortion = NONE
[... 155 lines cut ...]
  480	x 360	@ 30Hz	Y8	56.2 x 43.7 degrees, distortion = NONE
  480	x 360	@ 30Hz	Y16	56.2 x 43.7 degrees, distortion = NONE
  332	x 252	@ 90Hz	Y8	57.3 x 45 degrees, distortion = NONE
  332	x 252	@ 90Hz	Y16	57.3 x 45 degrees, distortion = NONE
  332	x 252	@ 60Hz	Y8	57.3 x 45 degrees, distortion = NONE
  332	x 252	@ 60Hz	Y16	57.3 x 45 degrees, distortion = NONE
  332	x 252	@ 30Hz	Y8	57.3 x 45 degrees, distortion = NONE
  332	x 252	@ 30Hz	Y16	57.3 x 45 degrees, distortion = NONE
  320	x 240	@ 90Hz	Y8	55.5 x 43.1 degrees, distortion = NONE
  320	x 240	@ 90Hz	Y16	55.5 x 43.1 degrees, distortion = NONE
[... cut ...]
  320	x 240	@ 30Hz	Y16	55.5 x 43.1 degrees, distortion = NONE

sometimes I get another results
rs.error: Failed to get firmware version
rs.error: Couldn't query adapter board / motion module FW version!
rs.warn: Motion module capabilities were disabled due to failure to acquire intrinsic
Device 0 - Intel RealSense ZR300:
 Serial number: 3511803652
 Firmware version: 2.0.71.28
 USB Port ID: 2-1-2
RealSense error calling rs_get_device_info(device:0x17bdb60, info:ADAPTER_BOARD_FIRMWARE_VERSION):
    selected camera info is not supported for this camera!

jixiedaima, 2017-12-26:
I run the driver following the two steps:
１．　https://github.com/IntelRealSense/realsense_sdk_zr300  相应的ＳＤＫ
２．　https://github.com/IntelRealSense/librealsense

icarpis, 2018-01-25:
Could you please try with another USB cable or even plug the camera to another USB port?

jkira33, 2018-02-22:
I have a same error like this. And I tried to plug the camera to other USB ports. It works sometimes. How to solve this problem  fundamentally？ it only relate to hardware or both?

jkira33, 2018-02-22:
I tried to launch launch the rgbd.launch many times and get an idea. Perhaps the error is related to the hardware. In my case, my USB ports are bad connect.

jixiedaima, 2018-02-23:
I think my driver is not installed correctly!

jlgarcia75, 2018-04-06:
Hello @jixiedaima,

Note that the current master librealsense branch does not support ZR300. You must use the legacy branch in Github: https://github.com/IntelRealSense/librealsense/tree/v1.12.1. 

Note that the ZR300 camera and all associated software is discontinued so support is limited.

Regards,
Jesus
Intel Customer Support

closed 2018-10-15 by dorodnic