realsenseai/realsense-ros #292: Camera Firmware Difference leads to No Camera Detected

markdole, 2018-01-17:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | 16.04                |
| Kernel                | ?-?-?                |
| ROS                   | kinetic                    |
| ROS RealSense         | 1.8.0                |
| librealsense          | 1.12.12               |
| Camera Type-Firmware  | W-1.0.71.16            |

### Expected Behavior
~$ roslaunch realsense_camera r200_nodelet_default.launch

### Actual Behavior

[ INFO] [1516179284.268959751]: /camera/driver - Detected the following camera:
				- Serial No: 2511006063, USB Port ID: 2-2-2, Name: Intel RealSense R200, Camera FW: 1.0.71.06
[ WARN] [1516179284.269090884]: /camera/driver - Detected unvalidated firmware:
				- 2511006063's current camera firmware is 1.0.71.06, Validated camera firmware is 1.0.72.06
[ INFO] [1516179284.269198231]: /camera/driver - Connecting to camera with Serial No: 2511006063, USB Port ID: 2-2-2
[ INFO] [1516179318.555551245]: /camera/driver - Setting static camera options
[ INFO] [1516179318.781861388]: /camera/driver - Enabling Depth in manual mode
[ERROR] [1516179318.782024833]: /camera/driver - Error calling rs_device_supports_option ( device:0x9ecf60, option:R200_AUTO_EXPOSURE_RIGHT_EDGE ): 
UVCIOC_CTRL_QUERY:UVC_GET_CUR error 2, No such file or directory 

[FATAL] [1516179319.549251801]: Failed to load nodelet '/camera/driver` of type `realsense_camera/R200Nodelet` to manager `nodelet_manager'
[camera/driver-2] process has died [pid 5896, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense_camera/R200Nodelet nodelet_manager depth:=depth color:=color ir:=ir ir2:=ir2 fisheye:=fisheye imu:=imu __name:=driver __log:=/home/ma/.ros/log/5c4aa346-fb58-11e7-9a38-d4bed9d3a2e0/camera-driver-2.log].
log file: /home/ma/.ros/log/5c4aa346-fb58-11e7-9a38-d4bed9d3a2e0/camera-driver-2*.log
[camera/nodelet_manager-1] process has died [pid 5895, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=nodelet_manager __log:=/home/ma/.ros/log/5c4aa346-fb58-11e7-9a38-d4bed9d3a2e0/camera-nodelet_manager-1.log].
log file: /home/ma/.ros/log/5c4aa346-fb58-11e7-9a38-d4bed9d3a2e0/camera-nodelet_manager-1*.log

### Steps to Reproduce

icarpis, 2018-01-17:
Hello @markdole, could you check please if the camera can streaming using LibRealSense capture tools by opening a Terminal and typing `cpp-config-ui` or `cpp-capture`?

markdole, 2018-01-18:
sorry, I typed and it said ' No command found'

icarpis, 2018-01-18:
Have you installed the [realsense_camera node](http://wiki.ros.org/realsense_camera) from [sources](http://wiki.ros.org/realsense_camera/Tutorials/Building_from_Sources) or from [Debian Package](http://wiki.ros.org/realsense_camera#ROS_Debian_Package)?

Please try to install LibRealSense following [these](http://wiki.ros.org/librealsense) instructions and try again to streaming with one of the RealSense capture tools.

markdole, 2018-01-18:
THANK U! I reinstalled the librealsense  and realsense_camera, the warning is still there ,but I can finally accept images from waffle by running 'roslaunch  realsense_camera r200_nodelet_rgbd.launch'

closed 2018-01-24 by icarpis