realsenseai/realsense-ros #297: Red and Blue RGB values swapped?

mchan60, 2018-01-24:
### System Configuration

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 16.04.3 LTS |
| Kernel                | 4.10.0-42-generic |
| ROS                   | kinetic |
| ROS RealSense         | 2.0.1 |
| librealsense          | 2.9.0 |
| Camera Type-Firmware  | Intel RealSense 435 |

### Expected Behavior
Red pixels are Red and Blue pixels are Blue

### Actual Behavior
Red pixels are Blue and vice versa

### Steps to Reproduce
launch rs_camera
open rviz
add an image that uses the /camera/color/image_raw topic

icarpis, 2018-01-25:
Hello @mchan60, could you please provide a picture that illustrate the issue?
I captured a [color bands](https://www.back2gaming.com/wp-content/uploads/2017/04/colorbands.png) image and I got the following output
[image]

mchan60, 2018-01-26:
@icarpis here is a screenshot of the same color bands image
[image]

icarpis, 2018-01-28:
Do you see the same issue also with the [realsense-viewer](https://github.com/IntelRealSense/librealsense/tree/master/tools/realsense-viewer) tool?
This tool is a part of Librealsense SDK and should be installed on your machine. either if you installed it using [Debian Package](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) or from [sources](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md).
Just open a Terminal and type `realsense-viewer`, then try to streaming color with RGB8 format.

Another thing that we can do is to remove RViz from the formula by trying to subscribe color topic using `rqt_image_view`. You can do it by typing `rqt_image_view` and choosing the `/camera/color/image_raw` topic.

mchan60, 2018-01-29:
After pulling the most recent commit the issue appears to have disappeared, so its possible that there was some setting that was accidentally changed causing this issue.

closed 2018-01-29 by mchan60

chris-wojciechowski, 2018-08-13:
I am experiencing the same issue, but it's not visible in the realsense-viewer, only when pulling frames from the camera in my software. I've updated the SDK to the latest version but this didn't seem to make any difference.

priyanandanaren, 2019-02-15:
I am experiencing the same issue. Can anyone has solution for this issue?

wangxian4423, 2019-11-20:
> I am experiencing the same issue. Can anyone has solution for this issue?

Well,just change the red channel and blue channel and everything works well!

doronhi, 2019-11-27:
@priyanandanaren , are experiencing this when subscribing on /camera/color/image_raw topic?
Can you share more parameters so w'll be able to reproduce? What camera, OS, realsense2_camera version, librealsense2 version? viewing application (rqt_image_view?)

priyanandanaren, 2019-11-27:
@doronhi - It works as per @wangxian4423 comments