realsenseai/realsense-ros #299: catkin_make error

shiyujin0, 2018-01-26:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 16.04.3 LTS             |
| Kernel                | 4.13.0-31-generic               |
| ROS                   | kinetic           |
| ROS RealSense         | 2.0.1             |
| librealsense          | 2.9.0              |
| Camera Type-Firmware  |Intel Realsense 415            |

---
When catkin_make, I got an error like this:
[ 92%] Building CXX object realsense-2.0.1/realsense_ros_camera/CMakeFiles/realsense_ros_camera.dir/src/realsense_camera_node.cpp.o
[ 92%] Built target realsense_ros_camera_generate_messages_py
[ 92%] Built target realsense_ros_camera_generate_messages_eus
Scanning dependencies of target realsense_ros_camera_generate_messages
[ 92%] Built target realsense_ros_camera_generate_messages
/home/mscnsf/catkin_ws/src/realsense-2.0.1/realsense_ros_camera/src/realsense_camera_node.cpp: In member function ‘realsense_ros_camera::IMUInfo realsense_ros_camera::RealSenseCameraNodelet::getImuInfo(const stream_index_pair&)’:
/home/mscnsf/catkin_ws/src/realsense-2.0.1/realsense_ros_camera/src/realsense_camera_node.cpp:993:58: error: ‘class rs2::sensor’ has no member named ‘get_motion_intrinsics’
             auto imuIntrinsics = _sensors[stream_index]->get_motion_intrinsics(
                                                          ^
realsense-2.0.1/realsense_ros_camera/CMakeFiles/realsense_ros_camera.dir/build.make:62: recipe for target 'realsense-2.0.1/realsense_ros_camera/CMakeFiles/realsense_ros_camera.dir/src/realsense_camera_node.cpp.o' failed
make[2]: *** [realsense-2.0.1/realsense_ros_camera/CMakeFiles/realsense_ros_camera.dir/src/realsense_camera_node.cpp.o] Error 1
CMakeFiles/Makefile2:1666: recipe for target 'realsense-2.0.1/realsense_ros_camera/CMakeFiles/realsense_ros_camera.dir/all' failed
make[1]: *** [realsense-2.0.1/realsense_ros_camera/CMakeFiles/realsense_ros_camera.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

closed 2018-01-27 by shiyujin0

ipa-jcl-hy, 2018-01-31:
Hi, 
can i ask how have you solved this problem. I get the same error when i do it today.

icarpis, 2018-01-31:
Hello @ipa-jcl-hy, if you have built the ROS wrapper from the [development](https://github.com/intel-ros/realsense/tree/development) branch, please install [RealSense 2.9.1](https://github.com/IntelRealSense/librealsense/tree/v2.9.1), else, each [ROS wrapper release](https://github.com/intel-ros/realsense/releases) has its appropriate [RealSense version](https://github.com/IntelRealSense/librealsense/releases).