realsenseai/realsense-ros #301: intelrealsenseD415（Given depth profile is not supported by current device!）

ccempathy, 2018-01-31:
I have installed Intel® RealSense™ SDK 2.0 (build 2.9.1) successfully and there is no problem installing ROS Wrapper for Intel® RealSense™ Devices.But there existed a question that "Given depth profile is not supported by current device!" when I was testing. I have been confused about the question for a long time .Look forward to your help .Thanks a lot!

[image]

icarpis, 2018-01-31:
Hello @ccempathy, could you please provide the whole output?
Also, which Launch file are you using? have you modified it?

ccempathy, 2018-01-31:
Hi @icarpis, I don't modified the launch file . I just make it according to source code and the whole output is below.

[image]
[image]

icarpis, 2018-01-31:
It looks that the camera has recognized as USB2 device (Device Product ID: 0AD6).
Have you connected it to USB3 port? if yes, could you please check again with another USB cable?

ccempathy, 2018-01-31:
I connect the camera to USB3 now and I try to make it connect the another port.But it still show the problem and I can open the camera by realsenseviwer .

icarpis, 2018-01-31:
please post the output of `lsusb` while the camera connected to USB3 port.

ccempathy, 2018-01-31:
Ok!Those are the two ports' results of my computer.

[image]

ccempathy, 2018-01-31:
@icarpis  I compiled the source code of ros wrapper (2.0.2) you uploaded just now. But when I ran the launch file it still showed the 2.0.1 of version.
[image]

icarpis, 2018-01-31:
@ccempathy, sorry for this issue, I've fixed it. please take it again from [here](https://github.com/intel-ros/realsense/tree/2.0.2).

icarpis, 2018-01-31:
Regarding the `lsusb` output, it seems that both ports aren't USB3 compliance.
You can check which USB version the USB controller is supported by typing `lsusb -t`.

ccempathy, 2018-02-01:
This is the output by typing lsusb -t below.The camera is connected to the port belong to bus 002 and it shows that the port on the bus 002 is USB 3.0 .
[image]

icarpis, 2018-02-01:
The 5 GB/s speed claimed by **Bus 02** means that it's a **USB3** interface and in other hand, the 480 MB/s speed claimed by **Bus 01**  means that it's a **USB2** interface .

ccempathy, 2018-02-02:
Now it can work succesfully.It is the 2.0 connecting line that result in the question.Thanks a lot for your help!

closed 2018-02-02 by icarpis

ccempathy, 2018-03-11:
@icarpis My camera is functioning correctly in the RealSense Viewer software that comes with the RealSense SDK 2.0.The detection picture is as follows.And i can run the order 'roslaunch realsense_ros_camera rs_camera.launch' to get the rgb picture.But When i run the order 'roslaunch realsense_ros_camera rs_rgbd.launch' to try to get pointcloud data. It fails and shows the mistake as follows. 
/home/tju/Pictures/Screenshot from 2018-03-06 20:40:56.png

ccempathy, 2018-03-11:
@icarpis   My camera is functioning correctly in the RealSense Viewer software  that comes with the RealSense SDK 2.0.The detection picture is as  follows.And i can run the order 'roslaunch realsense_ros_camera  rs_camera.launch' to get the rgb picture.But When i run the order  'roslaunch realsense_ros_camera rs_rgbd.launch' to try to get pointcloud  data. It fails and shows the mistake as follows.
--
[image]