realsenseai/realsense-ros #302: No point cloud?

j5207, 2018-02-01:
Hi all:
When I'm trying to roslaunch the point cloud by typing:
`roslaunch realsense_ros_camera rs_rgbd.launch`
The point cloud doesn't appear and nothing showing up.
The terminal looks like this:
[image]
Also the camera works well with realsense-viewer
I'm using Ubuntu 16.04 with Kinetic, Realsense D415, and librealsense-2.91 and realsense-2.02
It would be really helpful if anyone could help me out.

Thanks for time

icarpis, 2018-02-01:
Hello @j5207, running the `rs_rgbd.launch` by itself is not enough to visualize the RGBD. this Launch file is actually publishing the [ROS topics](http://wiki.ros.org/Topics) of RGBD point-cloud and the other available streams. One way to visualize the point-cloud is by using [RViz](http://wiki.ros.org/rviz) tool as the following:
1. Open Terminal and type `roslaunch realsense_ros_camera rs_rgbd.launch`
2. Open another Terminal and type `rviz`
3. Click on the `Add` button at the left bottom of the RViz window.
4. Click on the `By topic` tab.
5. Choose `pointcloud2` under `depth_registered` -> `points` and click `OK`

[image]

j5207, 2018-02-01:
@icarpis Thanks!

closed 2018-02-01 by j5207