realsenseai/realsense-ros #303: catkin_make link against librealsense2

a1lu, 2018-02-01:
Hello,
I tried to build librealsense2 and realsense_ros_camera in the same catkin workspace.
Since I did not install librealsense2 before I got an error during catkin_make in CMakeLists.txt:28:

```
find_package(realsense2)
if(NOT realsense2_FOUND)
    message(FATAL_ERROR "\n\n Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases\n\n")
endif()
```

I checked the available catkin variables, found librealsense2 in CATKIN_ORDERED_PACKAGES and created an workaround that works for me:

```
list(FIND CATKIN_ORDERED_PACKAGES "librealsense2" RealsenseInCatkin)
if(RealsenseInCatkin EQUAL -1)
    find_package(realsense2)
    if(NOT realsense2_FOUND)
        message(FATAL_ERROR "\n\n Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases\n\n")
    endif()
endif()
```

Can you reproduce this and is there a better solution to fix this?

Best regards

icarpis, 2018-02-01:
Hello @a1lu, it is recommended to install the RealSense SDK 2.0 (librealsense2) before installing the Intel ROS Wrapper (realsense_ros_camera). Please follow the [installation instructions](https://github.com/intel-ros/realsense#installation-instructions).
If you still want to use Catkin workspace to build also the SDK you need to change the catkin build priority. Maybe adding this line `add_dependencies(${PROJECT_NAME} librealsense2)` to `CMakeLists.txt` of the ROS wrapper will do the work but I've never tried it before.

a1lu, 2018-02-02:
Hello @icarpis ,
since librealsense has a [package.xml](https://github.com/IntelRealSense/librealsense/blob/master/package.xml) I thought it is supposed to build them in one workspace.
`add_dependencies()` does not work for me.
The workaround in the OP works for me. `find_package()` is run at configuration time (not build time) and obviously can't find librealsense2. If we skip this check catkin resolves the dependencies and builds the project.
Catkin independent builds still should resolve the dependencies through `find_package()`.
I am not a cmake/catkin expert though.

icarpis, 2018-02-05:
Most of the users are not using Catkin to compile the `librealsense2` together with the `ROS wrapper` so the problem is that if the `find_package()` will be removed and users will compile the `ROS wrapper` before `librealsense2`, they will get a bunch of compilation errors instead of getting a message saying that `Intel RealSense SDK 2.0 is missing`.

a1lu, 2018-02-06:
I dont want to remove `find_package()`. I just want to skip it if we build it with catkin and `librealsense2` is included in this workspace (member of `CATKIN_ORDERED_PACKAGES`), so it gets build by cmake and we can link against it.

stwirth, 2018-03-13:
I just ran into the same issue and @a1lu's workaround worked for me. It would be nice to get this fixed upstream!

scottnothing, 2018-03-16:
Perhaps related: when I tried `find_package(realsense2_camera REQUIRED)` in a separate project I encountered an error; cmake couldn't find `realsense2` specified in the `realsense2_camera` package. 

In `realsense2_camera/CMakeLists.txt`, `catkin_package(...` lists realsense2 as a CATKIN_DEPEND instead of a regular DEPEND, which was my issue.

a1lu, 2018-05-02:
The fix by @scottnothing does not work here. Did you change something else?

closed 2019-03-27 by RealSenseCustomerSupport