realsenseai/realsense-ros #304: Realsense R200 error: */camera/driver - Error calling rs_device_supports_option

RubensQRZ, 2018-02-01:
Hi, 

I get this error when trying to use the turtlebot3 waffle Realsense R200 camera using the command : "roslaunch turtlebot3_bringup turtlebot3_realsense.launch" or the command: "roslaunch realsense_camera r200_nodelet_default.launch"

**[ERROR] [1517504560.255244866]: **/camera/driver - Error calling rs_device_supports_option ( device:0x2369600, option:R200_AUTO_EXPOSURE_RIGHT_EDGE ): 
UVCIOC_CTRL_QUERY:UVC_GET_CUR error 2, No such file or directory**** 

I followed all the instructions available here (http://wiki.ros.org/librealsense) and here (http://emanual.robotis.com/docs/en/platform/turtlebot3/appendix_realsense/#installation)

This is what i get:

livre@livre-570x-DVT3:~$ roslaunch turtlebot3_bringup turtlebot3_realsense.launch
... logging to /home/livre/.ros/log/75eae038-0763-11e8-abc6-74e6e2d140c0/roslaunch-livre-570x-DVT3-15405.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.10.10.201:40131/

SUMMARY
========

PARAMETERS
 * /camera/driver/base_frame_id: camera_link
 * /camera/driver/camera_type: R200
 * /camera/driver/color_fps: 30
 * /camera/driver/color_frame_id: camera_rgb_frame
 * /camera/driver/color_optical_frame_id: camera_rgb_optica...
 * /camera/driver/depth_frame_id: camera_depth_frame
 * /camera/driver/depth_optical_frame_id: camera_depth_opti...
 * /camera/driver/fisheye_frame_id: camera_fisheye_frame
 * /camera/driver/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/driver/imu_frame_id: camera_imu_frame
 * /camera/driver/imu_optical_frame_id: camera_imu_optica...
 * /camera/driver/ir2_frame_id: camera_ir2_frame
 * /camera/driver/ir2_optical_frame_id: camera_ir2_optica...
 * /camera/driver/ir_frame_id: camera_ir_frame
 * /camera/driver/ir_optical_frame_id: camera_ir_optical...
 * /camera/driver/mode: manual
 * /camera/driver/serial_no: 
 * /camera/driver/usb_port_id: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.12

NODES
  /camera/
    driver (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)

ROS_MASTER_URI=http://10.10.18.2:11311

process[camera/nodelet_manager-1]: started with pid [15414]
process[camera/driver-2]: started with pid [15415]
[ INFO] [1517504505.609302519]: Initializing nodelet with 4 worker threads.
Intel RealSense F200_camera ; 2.60.0.0
Intel RealSense LR200_camera ; 2.0.71.18
Intel RealSense R200_camera ; 1.0.72.06
Intel RealSense SR300_camera ; 3.10.10.0
Intel RealSense ZR300_adapter ; 1.29.0.0
Intel RealSense ZR300_camera ; 2.0.71.28
Intel RealSense ZR300_motion_module ; 1.25.0.0
[ INFO] [1517504507.857575789]: /camera/driver - Detected the following camera:
				- Serial No: 2471011628, USB Port ID: 2-2-4, Name: Intel RealSense R200, Camera FW: 1.0.72.10
[ WARN] [1517504507.857724890]: /camera/driver - Detected unvalidated firmware:
				- 2471011628's current camera firmware is 1.0.72.10, Validated camera firmware is 1.0.72.06
[ INFO] [1517504507.857843712]: /camera/driver - Connecting to camera with Serial No: 2471011628, USB Port ID: 2-2-4
[ INFO] [1517504550.035593896]: /camera/driver - Setting static camera options
[ INFO] [1517504560.255059257]: /camera/driver - Enabling Depth in manual mode
**[ERROR] [1517504560.255244866]: **/camera/driver - Error calling rs_device_supports_option ( device:0x2369600, option:R200_AUTO_EXPOSURE_RIGHT_EDGE ): 
UVCIOC_CTRL_QUERY:UVC_GET_CUR error 2, No such file or directory**** 

**[FATAL] [1517504591.531324716]: Failed to load nodelet '/camera/driver` of type `realsense_camera/R200Nodelet` to manager `nodelet_manager'
[camera/driver-2] process has died [pid 15415, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense_camera/R200Nodelet nodelet_manager depth:=depth color:=color ir:=ir ir2:=ir2 fisheye:=fisheye imu:=imu __name:=driver __log:=/home/livre/.ros/log/75eae038-0763-11e8-abc6-74e6e2d140c0/camera-driver-2.log].
log file: /home/livre/.ros/log/75eae038-0763-11e8-abc6-74e6e2d140c0/camera-driver-2*.log
[camera/nodelet_manager-1] process has died [pid 15414, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=nodelet_manager __log:=/home/livre/.ros/log/75eae038-0763-11e8-abc6-74e6e2d140c0/camera-nodelet_manager-1.log].
log file: /home/livre/.ros/log/75eae038-0763-11e8-abc6-74e6e2d140c0/camera-nodelet_manager-1*.log**
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
livre@livre-570x-DVT3:~$ 
_____________________________________________________________
I have no idea what to do about it.

Thanks.

RubensQRZ, 2018-02-01:
Hi, 
I followed this instructions  "http://wiki.ros.org/realsense_camera/Tutorials/Building_from_Sources" and after that I used this command "sudo apt-get --reinstall install 'ros-*-librealsense'" And the error disappeared.

icarpis, 2018-02-04:
@RubensQRZ, please feel free to reopen if you encounter this issue again.

closed 2018-02-04 by icarpis