realsenseai/realsense-ros #305: publish_static_tf mixing axes

piotrpolatowski, 2018-02-05:
Hello,

I am curious why in `publish_static_tf` translation values are mixed, instead of assigning `x` or `trans.x`,`y` to `trans.y`, `z` to `trans.z`.

`void BaseRealSenseNode::publish_static_tf`
```
...
    msg.transform.translation.x = trans.z;
    msg.transform.translation.y = -trans.x;
    msg.transform.translation.z = -trans.y;
...
```
The first intuitive guess would be that `x` axis goes along image width (left to right), `y` along height (top to bottom) and z facing front. However the trick with swapping values changes that.

My initial guess was that flipping axes is a trick applying rotation, however if it was actually rotation the quaternion should also be rotated.

icarpis, 2018-02-15:
Hello @piotrpolatowski, the idea by doing this is to convert the axes system from the ROS standard axes system into optical frame axes system. see below 
[image]

You can find more details at the following link: [Coordinate Conventions](http://www.ros.org/reps/rep-0103.html#coordinate-frame-conventions)

piotrpolatowski, 2018-02-23:
@icarpis Thank you for the link with details. I misunderstood the original ROS Axes system.

closed 2018-02-23 by piotrpolatowski