realsenseai/realsense-ros #307: Set auto-exposure for IR cameras ZR300

waspinator, 2018-02-08:
When I launch the `roslaunch realsense_camera zr300_nodelet_rgbd.launch` file the IR cameras are over exposed to just white, and as a result there is no depth image. I get the same result if I run `cpp-config-ui` but I can check off the "LR auto exposure" box to get it to work. 

Is there a parameter in the launch file to set auto exposure?

jlgarcia75, 2018-02-08:
Hello @waspinator,

You can find the zr300_nodelet_rbgd.launch file here: https://github.com/intel-ros/realsense/blob/indigo-devel/realsense_camera/launch/zr300_nodelet_rgbd.launch.

I don't see a way to change the exposure in the launch file. I'll have to get help from engineering.

Regards,
Jesus
Intel Customer Support

jlgarcia75, 2018-04-06:
Hello @waspinator,

You can find all the configurable parameters here: http://wiki.ros.org/realsense_camera#ZR300_Dynamically_Reconfigurable_Parameters.

If a parameter is not in the launch file then the default value is used. Add any parameters you need and set accordingly.

Regards,
Jesus
Intel Customer Support

waspinator, 2018-04-07:
Thanks Jesus

closed 2018-04-07 by waspinator