realsenseai/realsense-ros #308: `rs_rgbd.launch` fails for D435 (ResourceNotFound: rgbd_launch)

moorage, 2018-02-08:
On ubuntu 16.04 with a fresh ROS kinetic install, `roslaunch realsense_ros_camera rs_camera.launch` starts up just fine, but `roslaunch realsense_ros_camera rs_camera.launch` fails with a Python error:

```
catkin_ws$ roslaunch realsense_ros_camera rs_rgbd.launch 
... logging to /home/myhome/.ros/log/aec32d7c-0d01-11e8-80a5-0800275780e7/roslaunch-mattmoore-VirtualBox-7697.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 306, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 748, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 720, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 654, in _recurse_load
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 684, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 588, in _include_tag
    inc_filename = self.resolve_args(tag.attributes['file'].value, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 370, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 383, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 151, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 197, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 203, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: rgbd_launch
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/mypath/Projects/myproject/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share
```
I followed the steps at https://github.com/intel-ros/realsense/blob/development/README.md .  I compiled `librealsense 2.9.1` as well as roswrapper for `realsense 2.0.2`

moorage, 2018-02-08:
It looks like running `sudo apt install ros-kinetic-rgbd-launch` helps at least

jlgarcia75, 2018-02-08:
Hi @moorage, do you still need help with this issue or did you resolve it?

Jesus

moorage, 2018-02-09:
hey @jlgarcia75 thanks for the reply.  I resolved that particular error, but still cannot view a depth cloud as described in the readme. 

For this image: https://user-images.githubusercontent.com/17433152/35396613-ddcb1d6c-01f5-11e8-8887-4debf178d0cc.gif the README suggests starting the depth camera on `roslaunch realsense_ros_camera rs_rgbd.launch`,  and then rosrun rviz... I can see the color raw_image, but when I select "PointCloud2", it never shows up.

see:
[image]

dpipemazo, 2018-02-09:
@moorage I ran into the same issue this morning after upgrading `librealsense` to `v2.10.0`. I downgraded to `v2.9.1` and resolved the issue. `v2.10.0` was only released yesterday it seems and might have broken something in the ROS package. Note that after downgrading `librealsense` you'll need to rebuild the `realsense_ros_camera` ROS package as well.

moorage, 2018-02-09:
@dpipemazo thanks!  How did you install v2.9.1 -- building from source?  I've been on v2.9.1 all along, and can't get the pointcloud2 to work.  Which issue are you running into exactly?

dpipemazo, 2018-02-09:
Yes, I installed both `librealsense` and this package from source. I followed the instructions hidden in `librealsense`'s docs that don't seem to be the main ubuntu installation instructions linked from their README [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md). I checked out their repo to tag `v2.9.1` before building. For this repo I'm building off of tag `v2.0.2`.

I have also noticed that I need to unplug and and re-plug the camera between runs of
```
roslaunch realsense_ros_camera rs_rgbd.launch
```
otherwise I run into the "Frames didn't arrived" issue. This is the same issue I was consistently seeing on `v2.10.0`.

Perhaps though, looking at your screenshot, there are other issues at play in your RVIZ configuration. To verify functionality I'd recommend setting the fixed frame to `camera_link` and then be sure that you select the `/camera/depth/points` topic after adding the PointCloud2 source in RVIZ.

moorage, 2018-02-13:
@dpipemazo thanks for the continued thoughts here.  What in my screenshot made you think there were issues at play?  Was it the global status error?

I tried reinstalling `v2.9.1`, as well as upping to `v2.10.1` and am still not able to see the pointcloud2.  I can see an updating flat image from the `/camera/color/image_raw` topic, but I can't get anything on pointcloud2.

I do receive some warnings when I start `rs_rgbd.launch`:

```
$ roslaunch realsense_ros_camera rs_rgbd.launch
... logging to /home/myhome/.ros/log/5693baca-1050-11e8-b094-0800275780e7/roslaunch-myhome-VirtualBox-1958.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://myhome-VirtualBox:46483/

SUMMARY
========

PARAMETERS
 * /camera/realsense_ros_camera/accel_fps: 1000
 * /camera/realsense_ros_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense_ros_camera/align_depth: True
 * /camera/realsense_ros_camera/camera_aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense_ros_camera/camera_aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense_ros_camera/camera_aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense_ros_camera/camera_aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense_ros_camera/color_fps: 30
 * /camera/realsense_ros_camera/color_height: 480
 * /camera/realsense_ros_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense_ros_camera/color_width: 640
 * /camera/realsense_ros_camera/depth_fps: 30
 * /camera/realsense_ros_camera/depth_height: 480
 * /camera/realsense_ros_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense_ros_camera/depth_width: 640
 * /camera/realsense_ros_camera/enable_color: True
 * /camera/realsense_ros_camera/enable_depth: True
 * /camera/realsense_ros_camera/enable_fisheye: True
 * /camera/realsense_ros_camera/enable_imu: True
 * /camera/realsense_ros_camera/enable_infra1: True
 * /camera/realsense_ros_camera/enable_infra2: True
 * /camera/realsense_ros_camera/enable_pointcloud: False
 * /camera/realsense_ros_camera/enable_sync: True
 * /camera/realsense_ros_camera/fisheye_fps: 30
 * /camera/realsense_ros_camera/fisheye_height: 480
 * /camera/realsense_ros_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense_ros_camera/fisheye_width: 640
 * /camera/realsense_ros_camera/gyro_fps: 1000
 * /camera/realsense_ros_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense_ros_camera/infra1_fps: 30
 * /camera/realsense_ros_camera/infra1_height: 480
 * /camera/realsense_ros_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense_ros_camera/infra1_width: 640
 * /camera/realsense_ros_camera/infra2_fps: 30
 * /camera/realsense_ros_camera/infra2_height: 480
 * /camera/realsense_ros_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense_ros_camera/infra2_width: 640
 * /camera/realsense_ros_camera/json_file_path: 
 * /camera/realsense_ros_camera/serial_no: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.12

NODES
  /camera/
    color_rectify_color (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    realsense_ros_camera (nodelet/nodelet)
    realsense_ros_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [1969]
[... 25 lines cut ...]
[ INFO] [1518479788.400679475]: RGB Camera was found.
[ INFO] [1518479788.400793451]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1518479788.400882955]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1518479788.400969633]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1518479788.401180409]: setupPublishers...
[ INFO] [1518479788.702614321]: setupStreams...
 12/02 15:56:28,704 WARNING [140299245901696] (sensor.cpp:313) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[ INFO] [1518479788.714456532]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1518479788.715251622]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1518479788.716244468]: infra2 stream is enabled - width: 640, height: 480, fps: 30
 12/02 15:56:28,719 WARNING [140299245901696] (backend-v4l2.cpp:1098) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 12/02 15:56:28,720 WARNING [140299245901696] (sensor.cpp:313) Unregistered Media formats : [ RW16 ]; Supported: [ ]
[ INFO] [1518479788.730132628]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1518479788.764152984]: publishStaticTransforms...
[ INFO] [1518479788.764470592]: RealSense Node Is Up!
 12/02 15:56:28,873 WARNING [140298478892800] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ WARN] [1518479788.880768368]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 12/02 15:56:29,238 WARNING [140298468026112] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
 12/02 15:56:29,346 ERROR [140299245901696] (types.h:178) xioctl(VIDIOC_S_CTRL) failed Last Error: Numerical result out of range
[ WARN] [1518479789.353421568]: Reconfigure callback failed with exception xioctl(VIDIOC_S_CTRL) failed Last Error: Numerical result out of range: 
[ WARN] [1518480044.606976943]: Messages of type 2 arrived out of order (will print only once)
[ WARN] [1518480044.607681886]: Messages of type 1 arrived out of order (will print only once)
```

Thanks so much for any continued help getting this to run with ROS!

moorage, 2018-02-13:
I figured out this particular issue ... I needed to set rviz “Fixed Frame” to be set to `camera_color_optical_frame`.

Unfortunately now my pointcloud freezes in rviz after 60 se

closed 2018-02-13 by moorage

reinzor, 2018-06-02:
Since the launch file depends on `rgbd_launch`, the package should at least have an `exec_depend`

yyxr75, 2019-12-04:
害