realsenseai/realsense-ros #309: OpenMP is missing!

afdaniele, 2018-02-09:
OpenMP is detected but the check at `realsense_ros_camera/CMakeLists.txt:21` (commit: ee4b529, release: 2.0.2) fails.
I can circumvent the problem just by avoiding that check, but this is just an hack. 

```
-- +++ processing catkin package: 'realsense_ros_camera'
-- ==> add_subdirectory(realsense-2.0.2/realsense_ros_camera)
-- Using these message generators: gencpp;genlisp;genpy
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Found OpenMP: -fopenmp  
CMake Error at realsense-2.0.2/realsense_ros_camera/CMakeLists.txt:21 (message):

   OpenMP is missing!

-- Configuring incomplete, errors occurred!
```

icarpis, 2018-02-11:
Hello @afdaniele, could you please elaborate which OS are you using and what is the compiler version (gcc -v)?

afdaniele, 2018-02-11:
Hi @icarpis , I'm using Ubuntu 14.04 (Kernel 4.4.0-111-generic) and ROS Indigo, GCC v5.4.1.
Let me know if there is any other information that I can provide.

icarpis, 2018-02-12:
Please check if OpenMP is installed on your machine using `ldconfig` by the following:
`$ ldconfig -p | grep gomp`
`libgomp.so.1 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libgomp.so.1`

Anyway, you can always compile the ROS wrapper without using OpenMP by the following line:
`catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release -DBUILD_WITH_OPENMP=False`

afdaniele, 2018-02-13:
I checked and `libgomp.so.1` is present in my system. I also found out that (related issue [#744](https://github.com/IntelRealSense/librealsense/issues/744)) building with OpenMP creates a library that uses all the 4 cores of my i5 4th-gen 2.7GHz CPU at 99% with a pointcloud published at ~4Hz. By building without OpenMP (as you suggested) I get ~20% CPU usage and pointcloud at ~12Hz. Thanks.

icarpis, 2018-02-15:
Feel free to open a new issue if you encounter any problem.

closed 2018-02-15 by icarpis