realsenseai/realsense-ros #311: OpenMP is missing!

kimkildae, 2018-02-12:
ubuntu 14.04.5 LTS 
ROS indigo
gcc version 5.4.1 20160904 (Ubuntu 5.4.1-2ubuntu1~14.04) 

i already install sdk 2.0(https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md)
i wnat install  realsense-2.0.2 but i have some problem

**CMake Error at realsense-2.0.2/realsense_ros_camera/CMakeLists.txt:21 (message):

   OpenMP is missing!

-- Configuring incomplete, errors occurred!**

icarpis, 2018-02-12:
Hello @kimkildae, please try to compile without OpenMP using the following line `catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release -DBUILD_WITH_OPENMP=false`

dorodnic, 2018-10-15:
This should be resolved (we no longer depend on OpenMP by default in librealsense)

closed 2018-10-15 by dorodnic