realsenseai/realsense-ros #313: No available methods to filter depth stream

poripsa, 2018-02-12:
None of the filters from realsense (library) can be applied without modifying the ros-wrapper

icarpis, 2018-02-13:
Hello @poripsa, we are planning to add the Depth filters in the next releases.

poripsa, 2018-02-13:
Hi @icarpis I will be trying to do it parallely with librealsense 2.10.0

icarpis, 2018-02-14:
Thank you @poripsa, that's would be great.
As a reference you can take a look at the implementation of [post-processing](https://github.com/IntelRealSense/librealsense/tree/development/examples/post-processing) example.

akifh, 2018-03-08:
@icarpis Any update on adding post-processing options to ROS wrapper? Or any guide on how and where to implement them?

piotrpolatowski, 2018-03-12:
@akifh quick way for most filters `base_realsense_node`. I have skipped decimation filter though, since it would require more changes because of resolution change.

1. declare filters in header.
```
        // Decalre filters
        std::shared_ptr<rs2::decimation_filter> decimation_filter;
        std::shared_ptr<rs2::spatial_filter> spatial_filter; // edge-preserving spatial smoothing
        std::shared_ptr<rs2::temporal_filter> temporal_filter; // reduces temporal noise
        std::shared_ptr<rs2::disparity_transform> depth_to_disparity;
        std::shared_ptr<rs2::disparity_transform> disparity_to_depth;
```
2. initialize in constructor
```
    decimation_filter = std::make_shared<rs2::decimation_filter>();
    temporal_filter = std::make_shared<rs2::temporal_filter>();
    spatial_filter = std::make_shared<rs2::spatial_filter>();
    depth_to_disparity = std::make_shared<rs2::disparity_transform>(true);
    disparity_to_depth = std::make_shared<rs2::disparity_transform>(false);
```
3. apply filters in `BaseRealSenseNode::setupStreams()` around line [563](https://github.com/intel-ros/realsense/blob/development/realsense2_camera/src/base_realsense_node.cpp#L563)                         
```
if (stream_type == RS2_STREAM_DEPTH)
{
  f = depth_to_disparity->process(f);
//                            f = decimation_filter->process(f);
  f = spatial_filter->process(f);
  f = temporal_filter->process(f);
  f = disparity_to_depth->process(f);
}
```

anleger92, 2018-03-14:
@piotrpolatowski  Thank you for your code :)
But when i insert the lines of code you posted I get errors. But this is probably because I just inserted them in the wrong order. I usually do not program in cpp. 
Could you just name the lines in which I have to insert the code you posted?
Would appreciate it.

johannesburg, 2018-07-07:
In case anyone needs a quick fix: I've added the filter code to a [forked repo](https://github.com/johannesburg/uw-realsense) of the realsense library and currently have three of the four filters added (temporal, spatial, and hole filling filter). Currently working on adding in the decimation filter.

closed 2019-03-26 by RealSenseCustomerSupport