realsenseai/realsense-ros #314: Unplug-Replug needed

afdaniele, 2018-02-12:
I am using a RealSense camera model D415 on Ubuntu 14.04 with ROS Indigo. I noticed that every time I stop the realsense node (i.e., run using `roslaunch realsense_ros_camera rs_rgbd.launch`) and I start it again, this is what I get. The node does not get frames from the camera. The only solution that I found so far is to unplug and replug the camera.

```
...
[ INFO] [1518476685.352818378]: setupPublishers...
[ INFO] [1518476685.369487645]: setupStreams...
[ INFO] [1518476685.378412925]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1518476685.378861850]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1518476685.379316215]: infra2 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1518476685.388006571]: color stream is enabled - width: 640, height: 480, fps: 30
 12/02 18:04:45,401 WARNING [140489646925568] (backend-v4l2.cpp:808) Empty frame has arrived.
 12/02 18:04:45,401 WARNING [140489646925568] (backend-v4l2.cpp:808) Empty frame has arrived.
 12/02 18:04:45,401 WARNING [140489646925568] (backend-v4l2.cpp:808) Empty frame has arrived.
 12/02 18:04:45,401 WARNING [140489646925568] (backend-v4l2.cpp:808) Empty frame has arrived.
[ INFO] [1518476685.405943127]: publishStaticTransforms...
[ INFO] [1518476685.406158396]: RealSense Node Is Up!
 12/02 18:04:45,406 WARNING [140489637914368] (backend-v4l2.cpp:808) Empty frame has arrived.
 12/02 18:04:45,406 WARNING [140489637914368] (backend-v4l2.cpp:808) Empty frame has arrived.
 12/02 18:04:45,406 WARNING [140489637914368] (backend-v4l2.cpp:808) Empty frame has arrived.
 12/02 18:04:45,406 WARNING [140489637914368] (backend-v4l2.cpp:808) Empty frame has arrived.
 12/02 18:04:50,401 WARNING [140489655318272] (backend-v4l2.cpp:825) Frames didn't arrived within 5 seconds
 12/02 18:04:50,406 WARNING [140489646925568] (backend-v4l2.cpp:825) Frames didn't arrived within 5 seconds
```

icarpis, 2018-02-13:
Hello @afdaniele, this is a known firmware bug and also reported by issues [1086](https://github.com/IntelRealSense/librealsense/issues/1086) and [1131](https://github.com/IntelRealSense/librealsense/issues/1131).
Currently we suggesting two options in order to workaround it. the first option is to unplug and replug the camera as you mentioned. the second option is to reset the camera by `hardware_reset` function each time before running the ROS wrapper.
I already implemented the second option [here](https://github.com/icarpis/realsense/tree/reset_dev), so I would be glad if you could test it.

afdaniele, 2018-02-13:
Thank you, I'll test it and keep you posted.

lakshmankumar1993, 2018-02-22:
@icarpis I came across the same unplug-replug problem and currently just unloading and reloading the uvcvideo kernel via modprobe works. I did have a look at the current code, I don't see any calls to _sens[stream].stop() and _sens[stream].close() in the destructor of base_realsense_node to stop and remove ownership of the stream. Do you think this is one of the reasons why the device segfaults (as seen in the kernel log message via dmesg) when trying to relaunch the ros node ?

icarpis, 2018-02-22:
@lakshmankumar1993, I think that adding `stop()` and `close()` to the dtor will not solved this issue since killing a process by `SIGINT` signal will not triggered the dtors. Moreover, I added the `stop` and `close` to the `SIGINT` callback handler and it's NOT solved the problem.
The `uvcvideo` driver should distinguish that the process has died and as a result stop steaming and release the resources.

lakshmankumar1993, 2018-02-22:
@icarpis Oh okay. Thanks for the reply. Btw, is there a way to use syncer with multiple queues instead of a single callback? I have multiple queues for each stream so that I can process them individually. So in case I wanted a synchronized queue, is there a way I can add multiple queues to this syncer ?

aangerma, 2018-02-27:
Hi @lakshmankumar1993, the frame callback gets a `frameset` which contains the synced frames, so you can go over the `frameset` and split each frame to its corresponding queue.

lakshmankumar1993, 2018-02-27:
@aangerma Yeah. That's what I ended up doing. Thanks.

atyshka, 2018-08-28:
@icarpis Would you consider a PR for this?

closed 2019-03-26 by RealSenseCustomerSupport