realsenseai/realsense-ros #315: d435 on v2.0.2 w/ librealsense v2.10.0:: `[ERROR] [1518537840.126603320]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).`

moorage, 2018-02-13:
I'm running realsense ros v `2.0.2` with librealsense `v2.10.0`.  When I boot up `roslaunch realsense_ros_camera rs_rgbd.launch`, I get a constant stream of the following error:

```
[ERROR] [1518537840.126603320]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
```

Over the course of running the camera for 30 seconds, occasionally, there's some of these interspersed:

```
[ERROR] [1518537921.717247204]: cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1518537921.726020775]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
```

As well as these, a couple times (not regularly): 

```
 13/02 08:04:04,348 WARNING [140517392402176] (backend-v4l2.cpp:825) Frames didn't arrived within 5 seconds
```

This also happens once:

```
 13/02 08:04:04,710 WARNING [140518176167808] (types.cpp:57) get_xu(id=11) failed! Last Error: Input/output error
[ WARN] [1518537844.712649470]: Reconfigure callback failed with exception get_xu(id=11) failed! Last Error: Input/output error:
```

When I `rosbag record` for these 30 seconds, I find that there is regular data being published to `/camera/color/image_raw`, but only once (at second 11) was there data published to `/camera/depth_registered/points`.

Does anyone have any thoughts as to why these issues could be happening?

icarpis, 2018-02-22:
Hello @moorage, please consider to go over [this](https://github.com/ros-drivers/usb_cam/issues/55) issue. it looks that there is a problem to record compressed depth image using `rosbag`.

Regarding the `Frames didn't arrived within 5 seconds` and `get_xu(id=11) failed! Last Error: Input/output error` warnings, please provide RealSense log in debug level by the following:
1. Type `export LRS_LOG_LEVEL=Debug`.
2. Reproduce the issue.
3. Post the output of `realsense_ros_camera` node.

basvroeg, 2018-07-16:
Facing similar issues here, when using a D435 on Ubuntu 14.04 with Ros Kinetic using Realsense ROS v2.0.3 and LibRealSense v2.10.2. 

Using the camera within the ROS setup all goes fine. As soon as rosbag record -a is used, error messages on image conversion start to appear. Narrowing this down to image sources leads to the notion that color images go fine, but for Infrared very frequent errors appear (8UC1 is not a color format) and for depth / aligned images they appear at a lower rate (16UC1) 

logging depth images also reduces frame rate from about 15 hz (as expected) to about 2-2.5 Hz. Logging more can reduce frame rate even further, to around 0.5-1 Hz.

Hope to have a look tomorrow into referenced issues, to see if this helps...

mingyokim, 2018-10-27:
I think the reason for the error is because some topics use proprietary message types from intel. If you just need the topic `/camera/depth_registered/points`, you can just run `rosbag record /camera/depth_registered/points`. Try to avoid using `-a` unless you really need all the topics.

RealSenseCustomerSupport, 2019-03-19:
----------------------------------------------

Close this issue for lack of activity. Please feel free to reopen if the issue is still relevant.

closed 2019-03-26 by RealSenseCustomerSupport

NovoG93, 2019-03-28:
Facing similar problems with ROS1 Kinetic Ubuntu 16.04 and D435.

jinfagang, 2019-04-11:
@mingyokim That is not a good idea. -a sometime is needed since we record points is useless.

Facing same issue on kinetic but not on melodic, it's actually a rosbag bug, seems already fixed in melodic rosbag. But for kinetic it still have this issue.

To fix it on kinetic, how to disable compressed image anyway? @basvroeg 
From link you provide can not really understand how to set interms of realsense launch file

basvroeg, 2019-04-18:
added this to the general launchfile specifying the namespace, just before the include of the camera nodelet:

`    <rosparam param="/realsense/infra1/image_rect_raw/disable_pub_plugins" >
      - 'image_transport/compressed'
      - 'image_transport/compressedDepth'
      - 'image_transport/theora'
    </rosparam>
    <rosparam param="/realsense/infra2/image_rect_raw/disable_pub_plugins" >
      - 'image_transport/compressed'
      - 'image_transport/compressedDepth'
      - 'image_transport/theora'
    </rosparam>
    <rosparam param="/realsense/depth/image_rect_raw/disable_pub_plugins" >
      - 'image_transport/compressed'
      - 'image_transport/compressedDepth'
      - 'image_transport/theora'
    </rosparam>
    <rosparam param="/realsense/color/image_raw/disable_pub_plugins" >
      - 'image_transport/compressed'
      - 'image_transport/compressedDepth'
      - 'image_transport/theora'
    </rosparam>`

CameronDevine, 2019-09-13:
I was able to get around this issue by using the `-x` option in `rosbag` to exclude the one topic that was causing issues. The full command to do this is,
```bash
rosbag record -a -x ".*(theora)"
```

Triocrossing, 2020-03-06:
Thanks for the idea of @CameronDevine, I modified it a little bit and now it works like a charm for me.

`rosbag record -a -O $1.bag -x "(.*)/theora | (.*)/theora/(.*) | (.*)/compressedDepth"`

"|" is used for excluding multiple topic/subtopics.

felixvd, 2020-04-08:
I get the error for all the depth image topics. I'm on 2.2.12 and Melodic. When I subscribe to the uncompressed depth image with `rqt_image_view`, I get this error:
`double free or corruption (!prev)`
`Aborted (core dumped)`

gonultasbu, 2020-09-25:
the correct regex format for me is the following

`rosbag record -o ~/ -a -x "(.*)theora(.*)|(.*)compressed(.*)"`

monajalal, 2023-11-15:
@gonultasbu 
using your command, no bag was saved

```
root@7a805700666e:/workspace# rosbag record -o ~/ -a -x "(.*)theora(.*)|(.*)compressed(.*)"
[ INFO] [1700066936.129318065]: Subscribing to /rosout_agg
[ INFO] [1700066936.131239891]: Recording to '/root/_2023-11-15-16-48-56.bag'.
[ INFO] [1700066936.134720777]: Subscribing to /rosout
[ INFO] [1700066936.138392535]: Subscribing to /camera/realsense2_camera_manager/bond
[ INFO] [1700066936.141379812]: Subscribing to /tf_static
[ INFO] [1700066936.142602206]: Subscribing to /tf
[ INFO] [1700066936.145413898]: Subscribing to /camera/stereo_module/parameter_descriptions
[ INFO] [1700066936.151472821]: Subscribing to /camera/stereo_module/parameter_updates
[ INFO] [1700066936.156670089]: Subscribing to /camera/rgb_camera/parameter_descriptions
[ INFO] [1700066936.159023293]: Subscribing to /camera/rgb_camera/parameter_updates
[ INFO] [1700066936.160995901]: Subscribing to /camera/align_to_color/parameter_descriptions
[ INFO] [1700066936.163198139]: Subscribing to /camera/align_to_color/parameter_updates
[ INFO] [1700066936.166001252]: Subscribing to /diagnostics
[ INFO] [1700066937.171336767]: Subscribing to /camera/depth/image_rect_raw
[ INFO] [1700066937.174795810]: Subscribing to /camera/depth/camera_info
[ INFO] [1700066937.176709788]: Subscribing to /camera/depth/metadata
[ INFO] [1700066937.179327005]: Subscribing to /camera/color/image_raw
[ INFO] [1700066937.180329311]: Subscribing to /camera/color/camera_info
[ INFO] [1700066937.181524309]: Subscribing to /camera/color/metadata
[ INFO] [1700066937.182494680]: Subscribing to /camera/aligned_depth_to_color/image_raw
[ INFO] [1700066937.185050478]: Subscribing to /camera/aligned_depth_to_color/camera_info
[ INFO] [1700066937.187028774]: Subscribing to /camera/extrinsics/depth_to_color
[ INFO] [1700066937.188047749]: Subscribing to /camera/stereo_module/auto_exposure_roi/parameter_descriptions
[ INFO] [1700066937.188988715]: Subscribing to /camera/stereo_module/auto_exposure_roi/parameter_updates
[ INFO] [1700066937.189890298]: Subscribing to /camera/rgb_camera/auto_exposure_roi/parameter_descriptions
[ INFO] [1700066937.190998247]: Subscribing to /camera/rgb_camera/auto_exposure_roi/parameter_updates
^
```

monajalal, 2023-11-15:
@gonultasbu 
however, using 
```
root@7a805700666e:/workspace# rosbag record -O mona_noetic.bag -a -x "(.*)theora(.*)|(.*)compressed(.*)"
[ INFO] [1700067159.717991764]: Subscribing to /rosout_agg
[ INFO] [1700067159.719413972]: Recording to 'mona_noetic.bag'.
[ INFO] [1700067159.720969980]: Subscribing to /rosout
[ INFO] [1700067159.722081049]: Subscribing to /camera/realsense2_camera_manager/bond
[ INFO] [1700067159.723081056]: Subscribing to /tf_static
[ INFO] [1700067159.724717571]: Subscribing to /tf
[ INFO] [1700067160.729423310]: Subscribing to /camera/stereo_module/parameter_descriptions
[ INFO] [1700067160.731729348]: Subscribing to /camera/stereo_module/parameter_updates
[ INFO] [1700067161.729881060]: Subscribing to /camera/rgb_camera/parameter_descriptions
[ INFO] [1700067161.732194890]: Subscribing to /camera/rgb_camera/parameter_updates
[ INFO] [1700067161.733336091]: Subscribing to /camera/align_to_color/parameter_descriptions
[ INFO] [1700067161.734822022]: Subscribing to /camera/align_to_color/parameter_updates
[ INFO] [1700067161.735833104]: Subscribing to /diagnostics
[ INFO] [1700067161.736798501]: Subscribing to /camera/depth/image_rect_raw
[ INFO] [1700067161.737794981]: Subscribing to /camera/depth/camera_info
[ INFO] [1700067161.740957394]: Subscribing to /camera/depth/metadata
[ INFO] [1700067161.743866104]: Subscribing to /camera/color/image_raw
[ INFO] [1700067161.745697492]: Subscribing to /camera/color/camera_info
[ INFO] [1700067161.746684795]: Subscribing to /camera/color/metadata
[ INFO] [1700067161.747610206]: Subscribing to /camera/aligned_depth_to_color/image_raw
[ INFO] [1700067161.748536983]: Subscribing to /camera/aligned_depth_to_color/camera_info
[ INFO] [1700067161.749415774]: Subscribing to /camera/extrinsics/depth_to_color
[ INFO] [1700067161.750556258]: Subscribing to /camera/stereo_module/auto_exposure_roi/parameter_descriptions
[ INFO] [1700067161.751413860]: Subscribing to /camera/stereo_module/auto_exposure_roi/parameter_updates
[ INFO] [1700067161.752286442]: Subscribing to /camera/rgb_camera/auto_exposure_roi/parameter_descriptions
[ INFO] [1700067161.753372010]: Subscribing to /camera/rgb_camera/auto_exposure_roi/parameter_updates

```

fixed the problem

gonultasbu, 2023-11-15:
@monajalal that's interesting, your terminal output declares that it is saved to ` '/root/_2023-11-15-16-48-56.bag'.`, was your rosbag emtpy?

CameronDevine, 2023-11-15:
@monajalal appears to be running the `rosbag` command as root, so there shouldn't be any permissions issues with it getting saved to `/root`.

gonultasbu, 2023-11-15:
Good catch. I'm guessing it may be hidden or if it is a shared resource system (something like a cluster, the computer name makes me think it is something like that) it may be getting deleted automatically. I would advise not running as root or simply saving it somewhere else where users have proper rw permissions as opposed to. `~/`.

monajalal, 2023-11-15:
@gonultasbu I am not sure what exactly was going on but fixed it and gave credit to you here https://robotics.stackexchange.com/a/105440/16739

CameronDevine, 2023-11-15:
@monajalal Looking at your stackexchange post I think the problem might have to do with a missing docker volume mounting.