realsenseai/realsense-ros #318: Auto-Exposure Not Consistently Working at System Init on D435

dpipemazo, 2018-02-22:
Running v2.0.2 of this package with v2.10.0 of `librealsense` and camera firmware `5.9.2.0`.

I noticed quite poor depth performance when viewing the Point Cloud in RVIZ after power cycling the D435 camera but no issue when using `realsense-viewer`. This caused the data in RVIZ to appear splotchy and inconsistent.

I then launched `rosrun rqt_reconfigure rqt_reconfigure` and noticed that the auto-exposure for the depth stream (first checkbox at the top of the `realsense` config menu) was indeed turned on (as indicated by the box being checked). However, unchecking this box and then re-checking it consistently fixed the issue indicating that auto-exposure at nodelet boot is not working. This might be an issue on the `librealsense` side, but I'm not too sure since it looked OK in `realsense-viewer`.

Mitchell-Hampton, 2018-02-23:
I am experiencing the same thing.

icarpis, 2018-02-26:
@Mitchell-Hampton, @dpipemazo, please post the output of the RealSense ROS node.

icarpis, 2018-02-28:
I managed to reproduce this issue and it looks like a firmware bug.
As a workaround you can physically replug the camera or call to [Hardware Reset](https://github.com/IntelRealSense/librealsense/blob/51e766389f75731be6d37a4f435b5c6b172609bc/tools/data-collect/rs-data-collect.cpp#L249) method.

dpipemazo, 2018-02-28:
@icarpis Thanks for looking into it! Sorry I didn't post the output of the node; got pulled away from working on it for a few days. Is there an issue filed on the firmware that I can follow for it? I am happy to file it if you'd like.

ammarhusain, 2018-03-10:
I just ran into this issue as well. The auto exposure works well after toggling it off & on in `rqt_reconfigure`. I am going to try calling `hardware_reset` and see if that helps. 

@icarpis I would be really interested in a firmware fix for this as well.

agrunnet, 2018-03-23:
Can you confirm that you are only seeing this on Linux AND ROS? I dont see this issue on Windows, but maybe I am not doing the right thing to reproduce. We need to figure out whether it is FW or SW.

reinzor, 2018-03-29:
We have experienced the same issue, this is the script we use to solve it now (as a workaround):
```
#!/bin/bash
rosrun dynamic_reconfigure dynparam set realsense2_camera_manager rs435_depth_enable_auto_exposure 0
rosrun dynamic_reconfigure dynparam set realsense2_camera_manager rs435_depth_enable_auto_exposure 1
```

This script can be put in the launch file where you start the node and nodelet manager.

ammarhusain, 2018-04-02:
@agrunnet : Yep, I am seeing it only on Linux & ROS. In the Realsense viewer on Windows it launches fine with depth auto exposure enabled.

ammarhusain, 2018-04-19:
Any updates on this issue?

fiorano10, 2018-05-09:
@reinzor, I'm on the latest version for the librealense, SDK, and so on. Your script doesn't seem to work, I still have to toggle it in rqt_reconfigure. Any suggestions?

reinzor, 2018-05-10:
@fiorano10 , this script performs exactly the same service calls than you are doing with the gui. Please check whether your node argument is correct:
```
Usage: dynparam set [options] node parameter value
```

reinzor, 2018-06-01:
Yes, re-check your nodelet manager name. The first argument (args attribute) has to point to the nodelet manager that is running your camera driver nodelet.

You can use the `rosnode` cli to obtain the names of the running nodes:
```
rosnode list
```

ZacharyTaylor, 2018-06-05:
I have had the same issue with the d415. I believe it is caused as setting the manual exposure causes the system to drop out of auto exposure mode, and the manual exposure is set on startup.

I fixed the issue on my system by adding

`if(config.rs415_depth_enable_auto_exposure){
    BaseD400Node::setParam(config, base_depth_enable_auto_exposure);
}`

After this line https://github.com/intel-ros/realsense/blob/development/realsense2_camera/src/rs415_node.cpp#L93. I expect the same fix would work for the d435 node

remod, 2018-06-07:
Thanks @ZacharyTaylor ! I can confirm it works for the d435.

lorenwel, 2018-06-07:
Thanks @ZacharyTaylor. I opened PR #392 implementing the proposed fix.

pumaking, 2018-07-16:
Hope to see this fixed soon!

Metalzero2, 2018-09-05:
Running the procedure that @dpipemazo mentioned in his original post, with the latest RealSense updates (that is librealsense version 2.16, Real Sense firmware version 05.10.03 and latest realsense2_camera [31a0e4d] ), it would seem that auto exposure is working on start up.

It would be interesting if others could conform this as well.

doronhi, 2018-10-29:
fixed in the latest commit. Will be available in the upcoming release, hopefully by tomorrow.

closed 2018-10-29 by doronhi