realsenseai/realsense-ros #319: Running D435 is causing other devices to error

Mitchell-Hampton, 2018-02-23:
I have an emlid reach gps module (https://emlid.com/reach/) connected using the nmea_natsat_driver package.

Whenever I run the rs_rgbd launch file the camera works perfectly but somehow causes the gps to lose nearly all signal instantly. Unplugging the camera or stopping the node allow the gps to return to normal nearly instantly.

Running the camera_viewer on windows along side the gps is fine.

dpipemazo, 2018-02-23:
This is likely going to be a power issue on your USB 5V output. Are you running from a laptop or a desktop? How many other devices are using the bus? I'd recommend trying to run the devices through a powered USB3 hub if you haven't already.

Mitchell-Hampton, 2018-02-24:
I haven't yet tried on a powered USB hub and will update after trying.

I don't see how it could be a power issue as I experienced this on two separate computers (one Laptop and on industrial PC). After first experiencing the issue I did think it could be a power issue and hence I removed a USB Radio which was using a large amount of power and a few simple devices however the problem still existed.

I also don't see how both can work fine simultaneously in windows if it is a power issue.

Mitchell-Hampton, 2018-02-26:
Tested with multiple Powered USB hubs. Issue still persists. It did successfully run both camera and GPS simultaneously a couple time by chance however I was unable to replicate. Used a separate powered USB hub for each device.

mhkabir, 2018-02-27:
It is probably USB3 RF interference. Pretty "normal", and I have experienced it on pretty much every RealSense camera. See this document from Intel : https://www.intel.com/content/www/us/en/io/universal-serial-bus/usb3-frequency-interference-paper.html.

My suggestion would be to increase the separation between the GPS antenna and the camera as much as possible. The most notorious radiation sources are the USB connectors on each side of your cable. You can also try shielding with copper tape, which works to varying degrees.

closed 2019-02-26 by doronhi