realsenseai/realsense-ros #320: Point cloud acquisition is slow (sensor_msgs/PointCloud2)

agari1229, 2018-02-25:
when i run
`roslaunch realsense_ros_camera rs_rgbd.launch
`
i can get pointcloud with RGB. But the publishing rate of `/camera/depth_registered/points` is 2~3Hz.

so, is there any ways to get pointcloud (sensor_msgs/PointCloud2) without RGB ?

icarpis, 2018-02-26:
Hello @agari1229, please post the output of `rostopic hz /camera/depth_registered/points` while running the ROS node using `rs_rgbd.launch`.

icarpis, 2018-02-27:
I created a simple application that calculates and prints the frequency of the `point cloud` topic.
I got the following results when running it with D435 device on Ubuntu 16.04 machine with Intel Core i7-3770 CPU and 16GB of RAM.
```
Avg Hz: 26.3158
Avg Hz: 25
Avg Hz: 26.3158
Avg Hz: 26.3158
Avg Hz: 26.3158
Avg Hz: 26.3158
Avg Hz: 26.3158
Avg Hz: 26.3158
Avg Hz: 26.3158
Avg Hz: 24.3902
Avg Hz: 24.3902
Avg Hz: 21.7391
Avg Hz: 25.641
Avg Hz: 25.641
Avg Hz: 25.641
Avg Hz: 25.641
Avg Hz: 25.641
Avg Hz: 25.641
Avg Hz: 25.641
```

Please compile and run the following sample code and post the output.
[CMakeLists.txt](https://github.com/intel-ros/realsense/files/1762550/CMakeLists.txt)
[main.cpp.txt](https://github.com/intel-ros/realsense/files/1762556/main.cpp.txt)

1. Download `main.cpp.txt` and `CMakeLists.txt` files.
2. Rename `main.cpp.txt` to `main.cpp`
3. Open `Terminal` and type `cmake`
4. Type `make`
5. Run the `rs_rgbd.launch` in other `Terminal` window
6. Run the compiled binary by typing `./realsense_ros_camera_example`

agari1229, 2018-02-28:
The result was
`Avg Hz: 37.037
Avg Hz: 26.3158
Avg Hz: 30.303
Avg Hz: 30.303
Avg Hz: 38.4615
Avg Hz: 30.303
Avg Hz: 29.4118
Avg Hz: 31.25
Avg Hz: 13.3333
Avg Hz: 26.3158
Avg Hz: 30.303
Avg Hz: 35.7143
Avg Hz: 33.3333
Avg Hz: 26.3158
Avg Hz: 35.7143
Avg Hz: 30.303
Avg Hz: 30.303
Avg Hz: 31.25
Avg Hz: 31.25
Avg Hz: 27.027
Avg Hz: 37.037
Avg Hz: 25.641
Avg Hz: 33.3333
Avg Hz: 33.3333
Avg Hz: 31.25
Avg Hz: 29.4118
Avg Hz: 32.2581
Avg Hz: 30.303
Avg Hz: 29.4118
Avg Hz: 30.303
Avg Hz: 32.2581
Avg Hz: 30.303
Avg Hz: 13.3333
Avg Hz: 38.4615
Avg Hz: 28.5714
Avg Hz: 33.3333
Avg Hz: 30.303
Avg Hz: 26.3158
Avg Hz: 37.037
Avg Hz: 29.4118
Avg Hz: 30.303
Avg Hz: 32.2581
Avg Hz: 30.303
Avg Hz: 27.027
Avg Hz: 37.037
Avg Hz: 27.027
Avg Hz: 31.25
Avg Hz: 34.4828
Avg Hz: 31.25
Avg Hz: 29.4118
Avg Hz: 30.303
`

agari1229, 2018-02-28:
I have another question about the frame rate. 
According to the data sheet, the maximum output of depth is 90 fps. But, when i run the launch file below, i couldn't get 90 fps. (I think that it is because the output of color does not correspond to 90 fps...)
So, what should i change my launch file to get 90 fps point cloud?

[image]

closed 2019-03-27 by RealSenseCustomerSupport