realsenseai/realsense-ros #321: three realsense cannot work together at the same time

xgdgxb, 2018-02-26:
When I use three realsense (two d415,one d435) together , it can only read two data. Do you know why?

icarpis, 2018-02-26:
@xgdgxb, could you please try to streaming with the same setup using `realsense-viewer` tool?

xgdgxb, 2018-02-26:
When I use realsense-viewer tool, it also cannot work. When  I open two cameras, it works well..but when I open three cameras, it cannot work. The warnning as follows:  

26/02 16:51:56,101 WARNING [140399520430784] (archive.cpp:325) All frames from stream 0x0x2159c90 are now released by the user
 26/02 16:52:19,926 WARNING [140398966064896] (backend-v4l2.cpp:809) Empty frame has arrived.
 26/02 16:52:19,927 WARNING [140398966064896] (backend-v4l2.cpp:809) Empty frame has arrived.
 26/02 16:52:19,927 WARNING [140398966064896] (backend-v4l2.cpp:809) Empty frame has arrived.
 26/02 16:52:19,927 WARNING [140398966064896] (backend-v4l2.cpp:809) Empty frame has arrived.
 26/02 16:52:31,757 WARNING [140398974457600] (backend-v4l2.cpp:809) Empty frame has arrived.
 26/02 16:52:31,757 WARNING [140398974457600] (backend-v4l2.cpp:809) Empty frame has arrived.
 26/02 16:52:31,757 WARNING [140398974457600] (backend-v4l2.cpp:809) Empty frame has arrived.
 26/02 16:52:31,757 WARNING [140398974457600] (backend-v4l2.cpp:809) Empty frame has arrived.

@icarpis

xgdgxb, 2018-02-28:
@icarpis   I used Thinkpad T450 which has three USB3.0 at the first time, it could not work when the depth_fps is set to 30, but it works well when the depth_fps is set to 15. When I use a better computer(msi), it works well when the depth_fps is set to 30. So it may be USB bandwidth issue，but also  CPU issue.

closed 2019-02-26 by doronhi