realsenseai/realsense-ros #323: Publish depth info of D415/D435

nouei, 2018-02-27:
params for camera controls can be changed with dynamic reconfigure but the wrapper do not publish them to use.

doronhi, 2019-02-26:
Hello,
So sorry for the year's delay..
A lot has changed regarding dynamic reconfigure since then.
If this still relevant, can you mention a control for example?
Thanks

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

nouei, 2019-03-07:
hi doronhi, Why do you want to ignore it?
Whern you open the driver from rqt_reconfigure  there are a lot of parameters which you can see them but you can not reach them from the published message

doronhi, 2019-03-10:
Hi @nouei 
I am sorry for the misunderstanding. I had absolutely no idea of ignoring anything. My question requested an additional info, an example, because I was not sure I understood your initial request.
 I am still not sure I follow and if you could give me an example of usage that you lack it would help me.
As I understand it now, there is some piece of information that you can reach by dynamic reconfigure method but it is not also published in a dedicated topic. Is that it?

nouei, 2019-03-10:
No problem, thank you for following up.
I meant for example in projection:
https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0
The wrapper doesn't set the **distance** between infrared and RGB cameras correctly. 
These parameters are exists in driver but I think there is a bug in wrapper.
Thanks

RealSenseCustomerSupport, 2019-04-08:
----------------------------------------------

Hi nouei,

Sorry for taking a while to come over to this question.

Would you please help provide more details on your last question/example? How did you validate the distance between infrared & RGB cameras?
>>>
The wrapper doesn't set the **distance** between infrared and RGB cameras correctly.
These parameters are exists in driver but I think there is a bug in wrapper.
<<<

Thanks!

RealSenseCustomerSupport, 2019-04-16:
----------------------------------------------

Hi nouei,

Wonder if there is any update? Or do you still need help on this one?
If nothing else is needed, this one will be closed out soon.

Thanks!

closed 2019-04-26 by RealSenseCustomerSupport