realsenseai/realsense-ros #327: Depth to Color alignment is incorrect.

jpapon, 2018-02-28:


jpapon, 2018-02-28:
A little confused as to why the alignment is off - it looks like you copied the code from https://github.com/IntelRealSense/librealsense/blob/master/src/proc/align.cpp

My only guess is that the version of transfer_pixel used here is incorrect.

bhaskara, 2018-03-06:
I'm also having this issue.  I compared running the old (v1) version of `realsense_camera` (which I know to work well) with the new one (`realsense_ros_camera`), and notice that the depth->rgb transform seems backward in the new one (and has had some rotation added too, which it didn't use to):

```
$ # This is with the old launch file running
$ tf_echo realsense_depth_frame realsense_rgb_frame
This is the pose of realsense_depth_frame wrt reference frame realsense_rgb_frame
- Translation: [0.005, -0.026, 0.001]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
            in RPY (radian) [0.000, -0.000, 0.000]
            in RPY (degree) [0.000, -0.000, 0.000]
$ # Here I shut down the launch file and started the realsense2 launch file instead
$ tf_echo camera_depth_frame camera_color_frame                                                              
This is the pose of camera_depth_frame wrt reference frame camera_color_frame
- Translation: [-0.005, 0.026, -0.001]
- Rotation: in Quaternion [-0.003, -0.001, 0.001, 1.000]
            in RPY (radian) [-0.005, -0.001, 0.002]
            in RPY (degree) [-0.311, -0.058, 0.110]
```

closed 2019-03-27 by RealSenseCustomerSupport