realsenseai/realsense-ros #330: Pointcloud at topic /camera/depth/color/points is not filtered

poripsa, 2018-03-02:
I have just had the chance to look at this. I see using rs_rgbd.launch  gives out a pointcloud at topic  /camera/depth_registered/points which looks better. However pointcloud at /camera/depth/color/points doesnt seem to be filtered and doesnt  look to be using filtered frame :
https://github.com/intel-ros/realsense/blob/fff166c82226f30eec662d6e0a2c7335b53dc30d/realsense2_camera/src/base_realsense_node.cpp#L1059

poripsa, 2018-03-13:
Well just by passing depth_frame to function publishITRTopic and using it as follows:

auto depth_data = reinterpret_cast<const uint16_t*>(depth_frame.get_data());
scaled_depth = static_cast<float>(*depth_data) * _depth_scale_meters;

filters the cloud, @icarpis let me know if I should submit a pull request.

RealSenseCustomerSupport, 2019-03-19:
----------------------------------------------

Close this issue for lack of activity. Please feel free to reopen if the issue is still relevant.

closed 2019-03-27 by RealSenseCustomerSupport

hongtaowu67, 2020-10-09:
I am a bit confused about the camera frame (camera_color_optical_frame or camera_depth_frame or others) for these two point clouds: /camera/depth/color/points and /camera/depth_registered/points. Could anyone elaborate a bit on this detail?Thanks.