realsenseai/realsense-ros #334: Failed to read busnum/devnum when using ROS

js333031, 2018-03-05:
I'm using librealsense (SDK 2.10.1) successfully on my Ubuntu 16.04 host. For example, I can use rs-depth-quality w/o issues multiple times, use rs-distance app (sample output below).  But when using ROS, there are "Failed to read busnum/devnum" errors.

I found an issue on github that had similar behavior but that issue might have been a platform issue, related to USB interface.  In my case, the librs demos work w/o fail. 

udev rules from SDK 2.10.1 are installed.

$  roslaunch realsense2_camera rs_aligned_depth.launch 
... logging to /home/jsshekha/.ros/log/18a8cb0c-208f-11e8-8dc8-501ac5f1b757/roslaunch-sp3-2884.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://sp3:37578/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 1000
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/camera_aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/camera_aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 1000
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_height: 480
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra2_width: 640
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/serial_no: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.12

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [2894]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 18a8cb0c-208f-11e8-8dc8-501ac5f1b757
process[rosout-1]: started with pid [2907]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [2924]
process[camera/realsense2_camera-3]: started with pid [2925]
[ INFO] [1520265960.553625524]: Initializing nodelet with 4 worker threads.
[ INFO] [1520265960.608479430]: RealSense ROS v2.0.2
[ INFO] [1520265960.608527461]: Running with LibRealSense v2.10.1
 05/03 11:06:00,609 WARNING [140419018057600] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
 05/03 11:06:00,609 WARNING [140419018057600] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device1
 05/03 11:06:00,609 WARNING [140419018057600] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device2
 05/03 11:06:00,609 WARNING [140419018057600] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device3
 05/03 11:06:00,609 WARNING [140419018057600] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device4
 05/03 11:06:00,609 WARNING [140419018057600] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device5
 05/03 11:06:00,717 WARNING [140419018057600] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
 05/03 11:06:00,717 WARNING [140419018057600] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device1
 05/03 11:06:00,717 WARNING [140419018057600] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device2
 05/03 11:06:00,718 WARNING [140419018057600] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device3
 05/03 11:06:00,718 WARNING [140419018057600] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device4
 05/03 11:06:00,718 WARNING [140419018057600] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device5
[ INFO] [1520265960.726441172]: getParameters...
[ INFO] [1520265960.794224941]: setupDevice...
[ INFO] [1520265960.794271742]: JSON file is not provided
[ INFO] [1520265960.794313213]: ROS Node Namespace: camera
[ INFO] [1520265960.794671381]: Device Name: Intel RealSense 410
[ INFO] [1520265960.794704256]: Device Serial No: 701512020789
[ INFO] [1520265960.794740346]: Device FW version: 05.08.15.00
[ INFO] [1520265960.794764171]: Device Product ID: 0x0AD2
[ INFO] [1520265960.794789775]: Enable PointCloud: Off
[ INFO] [1520265960.794821759]: Align Depth: On
[ INFO] [1520265960.794852854]: Sync Mode: On
[ INFO] [1520265960.794926258]: Device Sensors: 
[ INFO] [1520265960.794978480]: Stereo Module was found.
[ INFO] [1520265960.795035564]: (Color, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1520265960.795064428]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1520265960.795106408]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1520265960.795136069]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1520265960.795168965]: setupPublishers...
[ INFO] [1520265960.963806310]: setupStreams...
[ INFO] [1520265960.974164490]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1520265960.974671231]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1520265960.975129422]: infra2 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1520265960.990064269]: publishStaticTransforms...
[ INFO] [1520265960.990238002]: RealSense Node Is Up!
 05/03 11:06:00,991 WARNING [140418262800128] (backend-v4l2.cpp:808) Empty frame has arrived.
 05/03 11:06:00,991 WARNING [140418262800128] (backend-v4l2.cpp:808) Empty frame has arrived.
 05/03 11:06:00,991 WARNING [140418262800128] (backend-v4l2.cpp:808) Empty frame has arrived.
 05/03 11:06:00,991 WARNING [140418262800128] (backend-v4l2.cpp:808) Empty frame has arrived.
 05/03 11:06:05,719 WARNING [140418318595840] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0
 05/03 11:06:05,720 WARNING [140418318595840] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device1
 05/03 11:06:05,720 WARNING [140418318595840] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device2
 05/03 11:06:05,720 WARNING [140418318595840] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device3
 05/03 11:06:05,720 WARNING [140418318595840] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device4
 05/03 11:06:05,720 WARNING [140418318595840] (backend-hid.cpp:1091) Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device5
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[ INFO] [1520265966.844603966]: Interrupt Signal is received! Terminating RealSense Node...
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

$ rs-distance 
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense 410
    Serial number: 701512020789
    Firmware version: 05.08.15.00

The camera is facing an object 1.909 meters away.
The camera is facing an object 1.905 meters away.

js333031, 2018-03-05:
Looks like the issue could be due to having a R410 instead of R415. Can you please suggest a launch file example for R410 based pointcloud?

RealSenseCustomerSupport, 2019-03-19:
----------------------------------------------

Close this issue for lack of activity. Please feel free to reopen if the issue is still relevant.

closed 2019-03-27 by RealSenseCustomerSupport

Danguana, 2019-10-24:
Hi this same issue is occurring for me too.

24/10 12:31:49,144 WARNING [139799280408320] (backend-hid.cpp:1169) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.22.auto
 24/10 12:31:49,144 WARNING [139799280408320] (backend-hid.cpp:1169) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.10.auto
 24/10 12:31:49,144 WARNING [139799280408320] (backend-hid.cpp:1169) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
 24/10 12:31:49,144 WARNING [139799280408320] (backend-hid.cpp:1169) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.21.auto
 24/10 12:31:49,145 WARNING [139799280408320] (backend-hid.cpp:1169) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.2.auto
 24/10 12:31:49,145 WARNING [139799280408320] (backend-hid.cpp:1169) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.20.auto
 24/10 12:31:49,145 WARNING [139799280408320] (backend-hid.cpp:1169) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.1.auto
 24/10 12:31:49,145 WARNING [139799280408320] (backend-hid.cpp:1169) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.7.auto
 24/10 12:31:49,145 WARNING [139799280408320] (backend-hid.cpp:1169) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.13.auto
 24/10 12:31:49,145 WARNING [139799280408320] (backend-hid.cpp:1169) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.19.auto
 24/10 12:31:49,145 WARNING [139799280408320] (backend-hid.cpp:1169) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.6.auto
 24/10 12:31:49,145 WARNING [139799280408320] (backend-hid.cpp:1169) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.12.auto
 24/10 12:31:49,145 WARNING [139799280408320] (backend-hid.cpp:1169) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.18.auto
 24/10 12:31:49,145 WARNING [139799280408320] (backend-hid.cpp:1169) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.5.auto
 24/10 12:31:49,171 WARNING [139798937319168] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds

I am using Ubuntu 16.04 w/ kernel 4.15.0-65-generic and SDK 2.29.0

doronhi, 2019-10-24:
What kind of device do you use?

Danguana, 2019-10-24:
Thanks for your reply 

Name: Intel RealSense D435I, serial number: 841512070237, update serial number: 911623051115, firmware version: 05.11.15.00, USB type: 3.2

Danguana, 2019-10-24:
Using sudo modprobe -r hid_sensor_custom appears to solve the problem.

Danguana, 2019-10-25:
Today the problem has returned.