realsenseai/realsense-ros #339: 'rs_rgbd.launch' fails for D415

ccempathy, 2018-03-06:
when i run the order 'roslaunch realsense_ros_camera rs_rgbd.launch' , the mistake turn up follows.
[image]

ccempathy, 2018-03-07:
if you have free time. Could you please give me some help?@icarpis

icarpis, 2018-03-28:
Hello @ccempathy, maybe something went wrong with the installation of ROS distribution.
Please check if the other Launch files working well. try to run `rs_camera.launch`

ccempathy, 2018-03-28:
@afourteia I have changed it to false .But it still goes wrong and display the same mistake.

ccempathy, 2018-03-28:
@icarpis the distribution i installed is Indigo. And when i used kinect with the ROS distribution it went well.And re_camera.launch file could be run well.

ccempathy, 2018-03-28:
@icarpis @afourteia The mistake shows that the order 'eval' is wrong in the launch file named  rs_rgbd.launch.I tried
[image]
 to modify  the launch file.But eventually i failed.

xmojiao, 2018-04-19:
@icarpis @afourteia I have the same error, help please

pl95, 2019-02-12:
@ccempathy  Because eval is a new feature in Kinetic. Indigo won't recognize this new expression. 
http://wiki.ros.org/roslaunch/XML 
[image]

RealSenseCustomerSupport, 2019-03-26:
----------------------------------------------

Thank you @pl95.
Closing it for now. Please reopen if there is any further question.

closed 2019-04-03 by RealSenseCustomerSupport