realsenseai/realsense-ros #341: align r200 depth image to color image

duyilong, 2018-03-08:
Hi,
I followed the ROS tutorial of "realsense_camera" to [Change Camera Parameters from the Default Presets](http://wiki.ros.org/realsense_camera/Tutorials/change_camera_parameters).
I set the camera_type to R200.
set the mode to manual.
set the depth_width to the 640.
set the depth_height  to the 480.
set the color_width to the 640.
set the color_height  to the 480.
set the depth_fps to the 30.
set the color_fps  to the 30. 

and I run the modified launch file.
by `rostopic echo -p --nostr --noarr /camera/depth/image_raw`
I find the size of depth image is 640*480, same as color image.

does that mean the depth image I get is space-time aligned to color image.
I wish (u,v) in depth image and (u,v) in color image are One-to-one rightly correspondence.
But the datesheet said R200 depth stream doesn't support 640* 480, only 628* 468,480* 360 and 320* 240 are supported.
so how can I get space-time aligned depth and color images of the 640*480 resolution?

wish for your replies : )

jlgarcia75, 2018-04-06:
Hello @duyilong,

This article may help you: https://github.com/IntelRealSense/librealsense/blob/legacy/doc/projection.md

Regards,
Jesus
Intel Customer Support

RealSenseCustomerSupport, 2019-03-07:
[Please Ignore - RealSense system comment]

closed 2019-03-27 by RealSenseCustomerSupport

yuan0623, 2021-06-25:
hi, do you get the answer?

jinchenglee, 2022-05-06:
In replying the original question, I believe the depth and color image data are aligned and padded by legacy librealsense , according to https://github.com/IntelRealSense/librealsense/blob/legacy/doc/projection.md#r200, as quoted below.

R200
    ...
    2. Depth images are pixel aligned with the first infrared stream except for an optional 6 pixel offset

    Native depth images are six pixels smaller on all four sides, but are otherwise pixel aligned with infrared
    librealsense will pad the depth image or crop the infrared image if you request matching resolutions
    If you request matching resolutions, depth and infrared will use the exact same intrinsics  
    ...