realsenseai/realsense-ros #343: Error:No RealSense devices were found! Terminating RealSense Node...

shijiwensjw, 2018-03-16:
I have installed the librealsense as [http://wiki.ros.org/librealsense](url) showed and then I can use the driver correctly. But when I reboot my computer, It won't run correctly again, and the error like this:

[ERROR] [1521164177.352755241]: No RealSense devices were found! Terminating RealSense Node...
[FATAL] [1521164177.488413562]: Failed to load nodelet '/camera/color_rectify_color` of type `image_proc/rectify` to manager `realsense2_camera_manager'
[FATAL] [1521164177.488414412]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[FATAL] [1521164177.488460693]: Failed to load nodelet '/camera/points_xyzrgb_hw_registered` of type `depth_image_proc/point_cloud_xyzrgb` to manager `realsense2_camera_manager'
[camera/realsense2_camera-3] process has died [pid 3993, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/steven/.ros/log/677583fa-28ba-11e8-bc6b-14b31f0c1c81/camera-realsense2_camera-3.log].
log file: /home/steven/.ros/log/677583fa-28ba-11e8-bc6b-14b31f0c1c81/camera-realsense2_camera-3*.log
[camera/color_rectify_color-4] process has died [pid 3994, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load image_proc/rectify realsense2_camera_manager --no-bond image_mono:=color/image_raw image_rect:=color/image_rect_color __name:=color_rectify_color __log:=/home/steven/.ros/log/677583fa-28ba-11e8-bc6b-14b31f0c1c81/camera-color_rectify_color-4.log].
log file: /home/steven/.ros/log/677583fa-28ba-11e8-bc6b-14b31f0c1c81/camera-color_rectify_color-4*.log
[camera/points_xyzrgb_hw_registered-5] process has died [pid 3995, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb realsense2_camera_manager false rgb/image_rect_color:=color/image_rect_color rgb/camera_info:=color/camera_info depth_registered/image_rect:=aligned_depth_to_color/image_raw depth_registered/points:=depth_registered/points __name:=points_xyzrgb_hw_registered __log:=/home/steven/.ros/log/677583fa-28ba-11e8-bc6b-14b31f0c1c81/camera-points_xyzrgb_hw_registered-5.log].
log file: /home/steven/.ros/log/677583fa-28ba-11e8-bc6b-14b31f0c1c81/camera-points_xyzrgb_hw_registered-5*.log
[camera/realsense2_camera_manager-2] process has died [pid 3978, exit code 1, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/steven/.ros/log/677583fa-28ba-11e8-bc6b-14b31f0c1c81/camera-realsense2_camera_manager-2.log].
log file: /home/steven/.ros/log/677583fa-28ba-11e8-bc6b-14b31f0c1c81/camera-realsense2_camera_manager-2*.log

And I should reinstall the librealsense then it will work again. How can I solve this problem? Thanks!
By the way, I use ubuntu16.04 and kernel version is 4.13.0.

fungidick, 2018-03-16:
Try booting with kernel version 4.10. I had the same issue with 4.13.

shijiwensjw, 2018-03-16:
Thank you, I will have a try.

shijiwensjw, 2018-03-18:
When I use the kernel 4.10, I fond another problem like this:

> [ERROR] [1521352967.125342870]: An exception has been thrown: The requested pixel format 'INZI' is not natively supported by the Linux kernel and likely requires a patch!
> Alternatively please upgrade to kernel 4.12 or later. Last Error: Inappropriate ioctl for device
> [ERROR] [1521352967.125418213]: An exception has been thrown: The requested pixel format 'INZI' is not natively supported by the Linux kernel and likely requires a patch!
> Alternatively please upgrade to kernel 4.12 or later. Last Error: Inappropriate ioctl for device
> [FATAL] [1521352967.222348737]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
> [FATAL] [1521352967.222349928]: Failed to load nodelet '/camera/points_xyzrgb_hw_registered` of type `depth_image_proc/point_cloud_xyzrgb` to manager `realsense2_camera_manager'
> [FATAL] [1521352967.222349053]: Failed to load nodelet '/camera/color_rectify_color` of type `image_proc/rectify` to manager `realsense2_camera_manager'
> [camera/realsense2_camera_manager-2] process has died [pid 27043, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/steven/.ros/log/fad31b22-2a71-11e8-bcaf-14b31f0c1c81/camera-realsense2_camera_manager-2.log].
> log file: /home/steven/.ros/log/fad31b22-2a71-11e8-bcaf-14b31f0c1c81/camera-realsense2_camera_manager-2*.log
> [camera/realsense2_camera-3] process has died [pid 27044, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/steven/.ros/log/fad31b22-2a71-11e8-bcaf-14b31f0c1c81/camera-realsense2_camera-3.log].
> log file: /home/steven/.ros/log/fad31b22-2a71-11e8-bcaf-14b31f0c1c81/camera-realsense2_camera-3*.log
> [camera/color_rectify_color-4] process has died [pid 27045, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load image_proc/rectify realsense2_camera_manager --no-bond image_mono:=color/image_raw image_rect:=color/image_rect_color __name:=color_rectify_color __log:=/home/steven/.ros/log/fad31b22-2a71-11e8-bcaf-14b31f0c1c81/camera-color_rectify_color-4.log].
> log file: /home/steven/.ros/log/fad31b22-2a71-11e8-bcaf-14b31f0c1c81/camera-color_rectify_color-4*.log
> [camera/points_xyzrgb_hw_registered-5] process has died [pid 27046, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb realsense2_camera_manager false rgb/image_rect_color:=color/image_rect_color rgb/camera_info:=color/camera_info depth_registered/image_rect:=aligned_depth_to_color/image_raw depth_registered/points:=depth_registered/points __name:=points_xyzrgb_hw_registered __log:=/home/steven/.ros/log/fad31b22-2a71-11e8-bcaf-14b31f0c1c81/camera-points_xyzrgb_hw_registered-5.log].
> log file: /home/steven/.ros/log/fad31b22-2a71-11e8-bcaf-14b31f0c1c81/camera-points_xyzrgb_hw_registered-5*.log
> 
It seems the kernel version 4.10 dosen't support the ros package realsense2_camera, it must be 4.12 or later. But there is no problem when I use realsense_camera.

closed 2019-03-27 by RealSenseCustomerSupport