realsenseai/realsense-ros #344: Error reconnecting to camera after stopping launch file

bbferka, 2018-03-16:
What   |   Version
------------ | -------------
Camera | D435
Ubuntu | 16.04
ROS | Kinetic
librealsense | v2.10.0
realsense-ros | v2.0.2

Hi, I am using the D435 with depth alignment turned on. Every time I stop the launch file (Ctr+C) I have to unplug and re- plug my camera (I am using USB3). If I don't do that the second time I start the launch file there is no more depth data being streamed. The error message I get is:

```
[ INFO] [1521185359.194338525]: RealSense Node Is Up!
 16/03 08:29:19,196 WARNING [140028366399232] (backend-v4l2.cpp:808) Empty frame has arrived.
 16/03 08:29:19,196 WARNING [140028366399232] (backend-v4l2.cpp:808) Empty frame has arrived.
 16/03 08:29:19,196 WARNING [140028366399232] (backend-v4l2.cpp:808) Empty frame has arrived.
 16/03 08:29:19,196 WARNING [140028366399232] (backend-v4l2.cpp:808) Empty frame has arrived.
 16/03 08:29:24,192 WARNING [140028407703296] (backend-v4l2.cpp:825) Frames didn't arrived within 5 seconds
````

As I said reconnecting to the USB fixes this issue. I checked using the ``realsense-viewer``, and that one also says that it can not stream data from the stereo module. 

Any ideas what to problem could be?  Any help would be much appreciated. The camera is mounted on a robot so it is quite annoying having to reconnect all the time :)

mbjelonic, 2018-03-19:
We have the same issue and already the ZR300 had such kind of issues. This bug should be fixed asap.

akifh, 2018-03-21:
Related: https://github.com/intel-ros/realsense/issues/314

bbferka, 2018-03-21:
@akifh thank you! I did not see that issue. closing this one.

closed 2018-03-21 by bbferka