realsenseai/realsense-ros #345: Rosparam set not working

Nachmr, 2018-03-19:
Hi there!
I'm playing a bit with the realsense_ros_camera package of ROS (kinetic version) and my intel realsense d435 camera. One of the things that for me is interesting is to change some values of the color image via ros parameters. I tried to change it with `rosrun rqt_reconfigure rqt_reconfigure` command I can them and it actually works quite well.

The problems comes when I try to change this parameters directly in the prompt. If e.g. I try `rosparam set /camera/realsense_ros_camera_manager/rs435_color_exposure 1000` or directly via python code with `rospy.set_param('/camera/realsense_ros_camera_manager/rs435_color_exposure', 1000)` nothing happens.

It is a problem of my configuration or it actually doesn't work?

My softwares version:
librealsense-2.10.1
ROS kinetic
Ubuntu 16.04

stwirth, 2018-03-21:
You have to either set the parameter before launching the camera driver or you use the dynamic reconfigure API:
Via command line:
```
rosrun dynamic_reconfigure dynparam set my_camera_node_name rs435_color_exposure 1000
```
Via python API:
```
import dynamic_reconfigure
client = dynamic_reconfigure.client.Client("my_camera_node_name")
client.update_configuration({"rs435_color_exposure": 1000})
```
Here's [the tutorial on ros.org](http://wiki.ros.org/dynamic_reconfigure/Tutorials/UsingTheDynamicReconfigurePythonClient).

jonasgerne, 2019-01-15:
Hi, I updated to Realsense 2.17.1 today and now I'm facing the issue that dynamic_reconfigure doesn't work like it used to work. Using dynparam set via command line works fine:
`rosrun dynamic_reconfigure dynparam set /camera1/Stereo_Module/ "Enable Auto Exposure" True`

However, I want to set parameters using thy python API:
```
import dynamic_reconfigure  
client = dynamic_reconfigure.client.Client("/camera1/Stereo_Module/")  
client.update_configuration({"Enable Auto Exposure": "true"})  
```
Here, the update step doesn't work at all. Can someone help?

jonasgerne, 2019-01-15:
I'm sorry, my mistake was to test it in a python console without initializing a ROS node. Everything works fine now!

closed 2019-03-27 by RealSenseCustomerSupport