realsenseai/realsense-ros #346: A new tf attachment is not created for multiple D4XX cameras

AndyZe, 2018-03-19:
With the previous RealSense cameras, the tf link that the camera would attach to could be namespaced like this:

`  <arg name="camera"          default="camera_ee" />`

Where this camera would now attach to camera_ee_link. I haven't found a way to do this with the new D4XX. They seem to always attach to "camera_link" despite a namespace:

`  <arg name="namespace"           default="camera_ee"/>`  <== does not work

neomanic, 2018-03-20:
I'm just dealing with this exact problem, having dual D435s... and found a bit of a hack workaround. I have dual point-clouds in Rviz as of 5 minutes ago.

The parameter for the base `camera_link` is there in the source (see include/constants.h), but it's not exposed in the launch files. So I made my own launch file by copying `nodelet.launch.xml`, and altered it to expose the `base_frame_id` as a parameter, see  [r435.launch.txt](https://github.com/intel-ros/realsense/files/1828090/r435.launch.txt) which you can use as-is.

I also added a `tf_prefix` parameter that gets prefixed to all the frame ids so I could easily create a parent launch file as follows:
```
<launch>
  <arg name="rs_left_serial"    default="xxxxxx_redacted"/>
  <arg name="rs_right_serial"    default="xxxxxx_redacted"/>
  
  <group ns="rs_left">
    <include file="$(find my_robot_src)/launch/realsense_d435.launch"> 
      <arg name="serial_no"         value="$(arg rs_left_serial)"/>
      <arg name="tf_prefix"           default="rs_left_"/>
      <arg name="enable_pointcloud" value="true"/>
      <arg name="enable_sync"       value="false"/>
      <arg name="align_depth"       value="false"/>
      
      <arg name="depth_fps"                value="6"/>
      <arg name="infra1_fps"               value="6"/>
      <arg name="infra2_fps"               value="6"/>
      <arg name="color_fps"                value="6"/>
    </include>
  </group>

... do other camera here in group rs_right ...
</launch>
```

CxyCH, 2018-03-29:
Hello @neomanic 
        I try to use you r435.launch.txt to change the published  tf name like `camera_depth_frame` to `xxx_depth_frame`. But I still get the error info `(backend-v4l2.cpp 808) empth frame has arrived`. 
        I only change your code at line 68, 
`<node pkg="nodelet" type="nodelet" name="realsense_ros_camera" args="load realsense_ros_camera/RealSenseNodeFactory $(arg manager)">`, 
from `realsense_ros_camera` to `realsense2_camera` because this is the plugin name with this release package.  So can you help me please?  Thank you

neomanic, 2018-04-11:
Sorry for slow reply @CxyCH. I also get a similar error, which is fixed by unplugging and replugging the camera. It is apparently a firmware/hardware issue rather than a software one, as the latest datasheet says it's a known issue that will be resolved in future firmware. But your error could be different.

Creesy87, 2018-06-26:
Hello @neomanic I came to the same problem as stated here. I tried your solution by changing the nodelet.launch.xml and also the rs_multiple_devices.launch file

I still get a single camera_link where it looks like the back camera overwrites the front camera in the TF Tree.

Did you use other files to launch the cameras independently?

baumanta, 2018-08-08:
I have the same issue. The script @neomanic  suggested works for me. Now the frames have a prefix and can be distinguished. My problem is now that the pointclouds still have the generic name in the header. Do you have any idea how to to fix that? It would be nice if the new frame names would be used consistently such that conversions between frames are possible.

neomanic, 2018-09-20:
Seems like this is a problem that pops up regularly, when it should be fixed in the included `rs_multiple_devices.launch` and I'm still using my custom launch file to get this to work. If I fork the repo and fix it up, can you accept a pull request @doronhi, or is there too much hassle with contributor agreements?

baumanta, 2018-09-20:
I fixed this issue in PR  #471, it would be great if you could get that in @doronhi

neomanic, 2018-09-21:
Oh brilliant. Nice work @baumanta. I've commented over on the PR.

closed 2019-03-27 by RealSenseCustomerSupport