realsenseai/realsense-ros #347: No way to publish pointcloud without RGB

josephduchesne, 2018-03-20:
I would like to publish just the pointcloud (XYZ) data for a D435, and not waste CPU time aligning depth+color data, and bandwidth doubling the message size. 

Looking at the code, the only pointcloud2 publisher is publishing to /camera/depth/color/points. This seems like an oversight.

deadundead, 2018-03-29:
Try this launch file. It disables retreiving color camera images and all alignment processing, leaving only infra and depth.
[rs_depthcloud.launch.txt](https://github.com/intel-ros/realsense/files/1860225/rs_depthcloud.launch.txt)

josephduchesne, 2018-03-29:
I don't see how that could work since in base_realsense_node.cpp publishRgbToDepthPCTopic is skipped if `!is_frame_arrived.at(COLOR)` . If the color camera isn't enabled, there is no PointcCloud2 topic message published.

deadundead, 2018-03-31:
@josephduchesne What launch files are you using to start the node?
As I understand, they now by default use external ROS image processing pipeline (depth_image_proc) to process get RGBD cloud for better performance. That is cleary seen in the [rgbd.launch]( https://github.com/intel-ros/realsense/blob/859cb61dfdcd537d59322de460b3c51de9f6cc30/realsense2_camera/launch/rs_rgbd.launch#L171 ) file. 
``
Although, demo_pointcloud.launch, which I believe uses internal poinctloud processing, still works, they state on ROS wiki page:
> enable_pointcloud (bool, default: false)
>Specify if to enable or not the native pointcloud. By default, it is set to false due to performance issues. This option is depreciated in favor of the rgbd_launch pointcloud and will be removed in the near future.
The launch file I provided you with also uses depth_image_proc (but this time /point_cloud_xyz) to obtain pointcloud from depth map data. It also sets camera enable_color parameter to false. So it does just what you want - processing only depth map information. Hope this helps.

rupak-d, 2018-12-06:
I changed the launch file to params to load the point_cloud_xyz nodelet, as shown below-
args="load depth_image_proc/point_cloud_xyz $(arg manager) $(arg bond)" respawn="$(arg respawn)">

Now, the encodings for the color streams in the base_relasense_node.cpp show an error. Any idea what the encodings should be changed to?

doronhi, 2018-12-16:
The following command will publish a pointcloud without attaching rgb image:
roslaunch realsense2_camera rs_camera.launch filters:=pointcloud pointcloud_texture_stream:=RS2_STREAM_ANY enable_color:=false

closed 2019-01-07 by doronhi