realsenseai/realsense-ros #349: Given depth profile is not supported by current device!  (D415)

Daivesi, 2018-03-26:
when i run this a command "roslaunch realsense2_camera rs_camera.launch", it's said 

[ERROR] [1522032671.136005310]: Given depth profile is not supported by current device!
[FATAL] [1522032671.342244422]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera_manager-2] process has died [pid 31873, exit code 1, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/anbot/.ros/log/89a30946-30a0-11e8-8fb6-00e0c100028e/camera-realsense2_camera_manager-2.log].
log file: /home/anbot/.ros/log/89a30946-30a0-11e8-8fb6-00e0c100028e/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has died [pid 31874, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/anbot/.ros/log/89a30946-30a0-11e8-8fb6-00e0c100028e/camera-realsense2_camera-3.log].
log file: /home/anbot/.ros/log/89a30946-30a0-11e8-8fb6-00e0c100028e/camera-realsense2_camera-3*.log

ev-mp, 2018-03-28:
Hello @Daivesi,
One of the possible reasons is that the device was improperly enumerated as USB2 rather than USB3 device due to a variety of reasons : Power, Cables, Controller, Contacts. 

Can you share the output of `lsusb` when the error occur to confirm/remove this concern ?

You can also run `realsense-viewer` tool of Librealsense2 that displays this info in UI (along with FW version and additional useful info)

rpinnama, 2018-07-07:
I am also running into this issue with the D435. Here is the output
```
~ » lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 8086:0ad6 Intel Corp. 
Bus 001 Device 004: ID 17ef:602e Lenovo 
Bus 001 Device 005: ID 24f0:0142  
Bus 001 Device 003: ID 058f:6254 Alcor Micro Corp. USB Hub
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

tschmidt64, 2018-07-07:
I too am running into this issue with the D435. Here is my output:
```
~ » lsusb                                                       
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 8086:0ad6 Intel Corp. 
Bus 001 Device 004: ID 17ef:602e Lenovo 
Bus 001 Device 005: ID 24f0:0142  
Bus 001 Device 003: ID 058f:6254 Alcor Micro Corp. USB Hub
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

FENGZHANG123, 2018-07-09:
@Daivesi @tschmidt64  I have the same issue.Have you guys solved it now?

YuehChuan, 2018-07-14:
same issue here!

jinfagang, 2019-01-08:
@FYPBois @YuehChuan @FENGZHANG123 Same issue, Intel team should promote the error messages to be more clear.....................

ev-mp, 2019-01-08:
@tschmidt64, @jinfagang ,
>Bus 001 Device 006: ID 8086:0ad6 Intel Corp. 

Mean that the device is recognized as USB2 which does not support High-res/high-fps modes.
In case you're using USB2 port then you should probably adjust the launch files accordingly.

Note that there is no reliable way for the SDK to know in advance whether the port should be recognized as USB2 or USB3 - the enumeration is done automatically by host.

Retrieving the USB connection type, on the other hand,  is supported in Librealsense with `dev.get_info(RS2_CAMERA_INFO_USB_TYPE_DESCRIPTOR)`.
The USB connection type is also displayed in `rs-enumerate-devices` and `realsense-viewer` tools.

jinfagang, 2019-01-09:
@ev-map Well, I tried both USB2.0 and USB3.0, neither can work............ I think it is not this problem

ev-mp, 2019-01-09:
@jinfagang , 
Using USB3 port doesn't guarantee that the device will be enumerated as USB3 device - the resolution is affected by factors such as cable quality, the actual port power, USB host controller and even how fast that cable  is plugged-in (USB type-C is particularly prone to this).

Use the mentioned tools to establish that the host enumerates the device as USB3.
Once established, run and try those profiles with `realsense-viewer` for verification.
In case the viewer fails you can post the logs for review.

RealSenseCustomerSupport, 2019-04-16:
----------------------------------------------

Hi Daivesi,

Not sure what camera, ROS release, RS SDK release you had, I tested a D435 with RS 2.19.1 + ROS 2.2.3 today, and have no issue with running it.
Wodner if RS Viewer works well off your system? Have you ever successfully run ROS wrapper launches? Or can you please try again with latest ROS, RS SDK, and FW?

Thanks!

RealSenseCustomerSupport, 2019-04-25:
----------------------------------------------

Hi Daivesi,

Any update on this one? Do you still need help?
If there is nothing else, this one will be closed soon.

Thanks!

closed 2019-05-13 by RealSenseCustomerSupport