realsenseai/realsense-ros #353: [realsense2_camera] Device has to be re-connected when driver restarts on ubuntu server (16.04)

reinzor, 2018-03-29:
Hi,

We are having some troubles when restarting the driver on Ubuntu Linux 16.04 Server LTS:

1. Plug in the device
2. Start the driver (works fine)
3. Stop the driver
4. Start the driver (frames cannot be grabbed properly, often a depth or color stream missing)
5. Stop the driver
6. Unplug and plug the device
7. Start the driver (works fine)

We only have this issue on ubuntu server, not on ubuntu desktop (16.04). Here we can simple start and stop the driver and everything keeps working fine. 

Do you also encounter this issue? 
What are your thought on solving / debugging this?

Thanks

AndyZe, 2018-04-03:
I got that on the previous generation of RealSense cameras. If you're desperate, there is [this program](https://github.com/intel-ros/realsense/pull/239) that will find and reset the cameras for you. You would need to update it with the newer device #'s.

The new cameras work well but I'm only using Ubuntu Desktop (haven't tried Server).

reinzor, 2018-04-04:
@AndyZe , thanks! At least we have a way to automate it now instead of replugging ;)

GuidoBartoli, 2018-05-15:
I have the same problem with **Ubuntu Desktop 16.04** using latest **ROS 2.0.3 package** with latest **firmware 5.9.9.2** and **SDK 2.11**.

I'm working on a package relying on the official `realsense-camera` package to acquire data from D435, but every time I stop the ROS nodes, the camera cannot be restarted and needs to be manually replugged. I posted some details about the received error messages [here](https://github.com/intel-ros/realsense/issues/360#issuecomment-384758187).

I saw that some unofficial workarounds exist (like [here](https://github.com/icarpis/realsense/tree/reset_dev) and [here](https://github.com/intel-ros/realsense/pull/239)), but an official Intel update to the ROS package would be _much_ more appreciated.

DaxiongAI, 2018-05-22:
I have the same problem with Ubuntu Desktop 14.04 using latest ROS 2.0.3 package with SDK 2.11.

closed 2019-03-27 by RealSenseCustomerSupport