realsenseai/realsense-ros #360: restarting the rs_aligned_depth.launch creates issues

RajathSwaroop, 2018-04-06:
Hi,

I am using an Intel Realsense D415 camera with ROS and when I close a rs_aligned_depth.launch file process using keyboard interrupt and start it again, I get empty frames received error. Can you please provide a solution. Below is the output of my process in the terminal

[ INFO] [1522976513.064863759]: RGB Camera was found.
[ INFO] [1522976513.065003992]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1522976513.065031280]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1522976513.065041558]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1522976513.065063718]: setupPublishers...
[ INFO] [1522976513.222014094]: setupStreams...
[ INFO] [1522976513.229681868]: depth stream is enabled - width: 640, height: 480, fps: 60
[ INFO] [1522976513.230202205]: infra1 stream is enabled - width: 640, height: 480, fps: 60
[ INFO] [1522976513.230667241]: infra2 stream is enabled - width: 640, height: 480, fps: 60
[ INFO] [1522976513.240178910]: color stream is enabled - width: 640, height: 480, fps: 60
 05/04 18:01:53,304 WARNING [139643227035392] (backend-v4l2.cpp:808) Empty frame has arrived.
 05/04 18:01:53,304 WARNING [139643227035392] (backend-v4l2.cpp:808) Empty frame has arrived.
 05/04 18:01:53,304 WARNING [139643227035392] (backend-v4l2.cpp:808) Empty frame has arrived.
 05/04 18:01:53,304 WARNING [139643227035392] (backend-v4l2.cpp:808) Empty frame has arrived.
[ INFO] [1522976513.308964484]: publishStaticTransforms...
[ INFO] [1522976513.309395300]: RealSense Node Is Up!
 05/04 18:01:53,309 WARNING [139643144107776] (backend-v4l2.cpp:808) Empty frame has arrived.
 05/04 18:01:53,309 WARNING [139643144107776] (backend-v4l2.cpp:808) Empty frame has arrived.
 05/04 18:01:53,309 WARNING [139643144107776] (backend-v4l2.cpp:808) Empty frame has arrived.
 05/04 18:01:53,309 WARNING [139643144107776] (backend-v4l2.cpp:808) Empty frame has arrived.

josephduchesne, 2018-04-09:
I believe that this is caused by this: https://github.com/IntelRealSense/librealsense/issues/1086

which also appears to be DSO-8665 here: https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense-D400-Series-Spec-Update.pdf

There's a workaround branch by icarpis here that resets the camera on startup: 
https://github.com/icarpis/realsense/tree/reset_dev

GuidoBartoli, 2018-04-26:
My setup is ROS Kinetic on Ubuntu 16.04 with RealSense ROS v2.0.3, LibRealSense v2.10.3 and D435 device with 05.09.02.00 firmware.
I'm having the same issue every time I relaunch `roslaunch realsense2_camera rs_camera.launch` after interrupting it with usual CTRL+C.

I noted also that, after the ROS launch failed, this problem is not resolved neither by the official `realsense-viewer`, the streams are empty and these are the error messages on the console:

```
WARNING [140482218329856] (types.cpp:57) get_xu(id=11) failed! Last Error: Input/output error
WARNING [140482218329856] (types.cpp:57) get_xu(id=11) failed! Last Error: Input/output error
WARNING [140482218329856] (sensor.cpp:315) Unregistered Media formats : [ UYVY ]; Supported: [ ]
WARNING [140482218329856] (types.cpp:57) set_pu(id=3) failed! Last Error: Input/output error
WARNING [140482218329856] (sensor.cpp:694) Exception was thrown when inspecting properties of a sensor
WARNING [140482218329856] (types.cpp:57) set_pu(id=9) failed! Last Error: Input/output error
WARNING [140482218329856] (sensor.cpp:694) Exception was thrown when inspecting properties of a sensor
```

and this is shown in the graphical interface:

```
Invalid Value in rs2_get_option(options:0x7fc48c005250, option:Enable Auto Exposure):
get_xu(id=11) failed! Last Error: Input/output error
```

The only solution I found so far is disconnect and replug the device, which is a bit annoying since I use D435 only with ROS and my current working project involves starting/stopping capture many times.

Is this [unofficial branch](https://github.com/icarpis/realsense/tree/reset_dev) the only solution or Intel is working on an official update on firmware or ROS package?

Thanks,
Best regards

GuidoBartoli, 2018-04-26:
Just a small update on `realsense-viewer`.
Launching the tool with the camera just replugged (a fresh start, like it was launched for the first time), streams are working fine, but I always receive these warnings... is this a normal behavior, too?

```
WARNING (sensor.cpp:315) Unregistered Media formats : [ UYVY ]; Supported: [ ]
WARNING (types.cpp:57) set_pu(id=3) failed! Last Error: Input/output error
WARNING (sensor.cpp:694) Exception was thrown when inspecting properties of a sensor
WARNING (types.cpp:57) set_pu(id=9) failed! Last Error: Input/output error
WARNING (sensor.cpp:694) Exception was thrown when inspecting properties of a sensor
```

Additionally, comparing them with the ones received after the the ROS launch was interrupted (see my previous comment), it seems that the problem occurs when `get_xu(id=11)`.

Hope this can be of any help in solving this.

RajathSwaroop, 2018-04-26:
@josephduchesne  I will look into the links you mentioned. I have been using realsense with ROS package for the last 2 months and I suddenly started getting an error message if I run the ros wrapper for real-sense. Here is the error:

[camera/realsense2_camera_manager-1] process has died [pid 5159, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/rajath/.ros/log/12fc3a60-4338-11e8-9d35-b827ebf080fe/camera-realsense2_camera_manager-1.log].

It starts normally at first publishing the topics and ends up in this error after 30 seconds. Does anyone know the solution to this??

caiofis, 2018-06-27:
@RajathSwaroop How did you fix the problem using the camera over the network? My camera stop publishing after 30 seconds when I run using multiple computers.

RajathSwaroop, 2018-06-27:
@caiofis  Yes my issue got resolved. Most likely if you look into the error log file you might find a connection refused error. Check if you are able to ping the ROS master from all the other computers that need the feed. If you are able to ping then check for rostopic list (if it shows up all good). If you get to communicate to the ROS master and get the topic list then check if you have set ROS IP correct on the machine that you use to start the realsense ros node. Either you wont be able to ping across one machine or its ROS MASTER URI is not set correct (for the machine running realsense node) or ROS IP is not set correct.

closed 2019-03-27 by RealSenseCustomerSupport