realsenseai/realsense-ros #363: aligned_depth_to_color/camera_info publishes depth camera info instead of color

lorenwel, 2018-04-09:
Hi,

when publishing aligned depth data the corresponding camera_info topic publishes the camera matrix of the depth camera. I find this counter-intuitive and in my opinion this goes against standard ROS practices. Since the depth data is now aligned with a different camera with a different calibration matrix, the `aligned_depth_to_.../camera_info` should publish the info of whatever camera the depth is aligned to. 

Here's a quick example of an issue/nuisance I encountered because of this behavior. 
We have a Realsense D435 mounted on our quadrupedal robot. We collected data on flat and level ground, logging the `aligned_depth_to_color/image_raw` and `aligned_depth_to_color/camera_info` topics among others. When later playing the data back as a DepthCloud in Rviz, this is the depth data I was shown: 
[image]

Again, the robot was standing on flat and level ground, so this is not correct. The cause for this is that the depth to point cloud converter in RVIZ rightfully uses the camera_info topic corresponding to whichever image topic was selected. Luckily, I also recorded the color image stream + camera info and by manually remapping the color camera_info to the aligned_depth_to_color camera_info when playing back the ROSbag, the data was displayed correctly:
[image]

If I wouldn't have logged the color info as well all the data I recorded would have been useless. 

I therefore strongly suggest to publish the camera info of whichever camera the depth data is aligned to. 
I imagine that this behavior also holds for the aligned_depth_to_infra topics, where of course this is a non-issue since infra and depth camera matrices are the same.

I am happy to open a pull-request if the maintainers agree with me

icarpis, 2018-04-09:
Hello @lorenwel, thank you for bringing this issue to our attention and for the detailed information.
I agree with the findings and would be happy to get a PR.

lorenwel, 2018-04-10:
@icarpis Opened PR to fix this #364

icarpis, 2018-04-15:
PR was merged

closed 2018-04-15 by icarpis