realsenseai/realsense-ros #365: Installation Failing on Step 1 in Ubuntu 14.04

rkeatin3, 2018-04-12:
Following the directions [here](https://github.com/intel-ros/realsense):

Step 1.a. Suggests installing debian packages. When I get to librealsense2-utils, I get the following error:

`The following packages have unmet dependencies:
 librealsense2-utils : Depends: librealsense2 (= 2.10.3-0~realsense0.66) but it is not going to be installed
                       Depends: libglfw3 (>= 3.1) but it is not installable
E: Unable to correct problems, you have held broken packages.
`
Perhaps that option may only apply to 16.04...

Step 1.b. Suggests installing by following instructions [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages). When I run the script `./scripts/patch-realsense-ubuntu-xenial.sh
`, I get the error: 

`Handle uvcvideo :
Applying the patched module ... modprobe: ERROR: could not insert 'uvcvideo': Invalid argument
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
modprobe: ERROR: could not insert 'uvcvideo': Invalid argument`

Should I be using a specific kernel? I have had different errors returned when running the patch script with different kernels (4.4.0-109, 4.4.0-116, etc).

ev-mp, 2018-04-15:
Hello @rkeatin3 ,
The debian packages for LibRealSense are built and tested with Ubuntu 16.0.X LTS (currently kernels 4.4,4.10 and 4.13) 64bit. If you're running a different distribution/Ubuntu version then the packages may not be fully compatible. 

Patching the kernel with script is an alternative route (you cannot have the kernel driver patched both by the manual script and DKMS), and is not recommended as a general rule:
- Relatively high maintenance "cost" as you have to rebuild the kernel modules each time the kernel is upgraded.
- The patch requires to user to update the kernel to the latest version of the LTS line, e.g. for 4.4.- kernels 4.4.0-109, 4.4.0-112, and 4.4.0-119 you'll have to run 4.4.0-119, otherwise it is not guaranteed to succeed.

With all that said I'd suggest to stick to the debian packages if you're running Ubuntu LTS 64bit.
As for `libglfw3` error - please see the answer to a similar question on LibRealSense forum [#722](https://github.com/IntelRealSense/librealsense/issues/722).

rkeatin3, 2018-04-16:
To clarify, I'm on Ubuntu 14.04.

Anyway, I didn't mean to imply I had did both steps concurrently; I tried both separately.

I did use the script to install `glfw3`, but apt-get still complains that the dependency is missing when I try to install the LibRealSense packages. Maybe that's because it gets installed in `/usr/local/lib` instead of `/usr/lib`; I don't know apt-get manages dependencies with packages installed from source.

I tried using `apt-get download` to sidestep this problem, but that still requires me to use the argument `--ignore-depends libglfw3` when installing the deb with `dpkg`. That seemed okay, but once I had installed `librealsense2` and `librealsense2-utils`, `librealsense2-dkms` complained that it didn't have `dkms`. In trying to use apt-get to install it, I see that it _also_ needs `glfw3`. I decided to stop here because we're subverting the whole purpose of these package management tools, but I ended up having to go through the process of installing `dkms` just to remove `librealsense2-dkms` because post-removal script was failing without `dkms` installed. 

Perhaps I'm missing something here, but I don't think it's unreasonable to put the onus on you for more detailed installation instructions that work with the Ubuntu distributions that you support.

rkeatin3, 2018-04-17:
Is the installation of the original librealsense still supported? I would prefer to use that version anyway, as I have a bunch of code written by a coworker that uses the older library.

wucm1994, 2018-07-19:
Hi! I also encountered this problem and I have solved it in this way:

1. Download this install script for trusty: [libfreenect2](https://github.com/OpenKinect/libfreenect2/blob/master/depends/download_debs_trusty.sh)
2. Enter to the ./debs folder, use `sudo dpkg -i libglfw3*deb`
3. ` sudo apt-get install -f`

Then the unmet dependency problem can be solved and you can install librealsense2-utils on your ubuntu 14.04.

closed 2019-03-27 by RealSenseCustomerSupport