realsenseai/realsense-ros #369: Compressed depth from D435 ROS Kinetic publishing a low FPS (approx 3Hz)

pirobot, 2018-04-20:
Hello,

We are using a D435 on ROS Kinetic, Ubuntu 16.04 with all the default parameters.  I compiled the development branch and it can see the camera and publish RGB and depth topics.  However, while the uncompressed depth topic /camera/depth/image_rect_raw is publishing at 30Hz, the compressed depth topic  /camera/depth/image_rect_raw/compressedDepth publishes at only 3Hz.

I have tried restarting the driver and the computer but the result is always the same.

jpapon, 2018-05-01:
I believe this is an issue with the way ROS compresses depth images (as a png). PNG compression is very slow - it wouldn't surprise me to learn you're only getting 3fps compressing a 1280x720 depth image to a png. You can try reducing the png compression level using dynamic reconfigure, but it's still not going to be very fast.

pirobot, 2018-05-02:
Thanks @jpapon.  However, I have never had this problem with any other RGB-D camera I have used with ROS including the original Kinect, Asus Xtion Pro Live and Stereo Labs ZED.  Also, I am running the camera at only 640x480 resolution.  Just now I tested my old Kinect under ROS Kinetic/Ubuntu 16.04 and I'm getting 30 fps for both RGB and depth images at 640x480 resolution.

jpapon, 2018-05-02:
Try adjusting lowering the quality of compressedDepth using dynamic reconfigure. You’ll see the frame rate of the compressedDepth topic increase.

In my experience, the png encoder is the bottleneck. Maybe at VGA res it’s okay, but certainly at 1280x720 it will kill you.

It doesn’t matter what camera you use, the compressedDepth is going to be slow. Though it’s possible other cameras have smoother depth maps which might be faster to compress?

pirobot, 2018-05-05:
OK thanks.  I've switched to a ZED stereo camera for now but I was able to confirm the effect of PNG compression level on FPS on the ZED so I'm guessing you are right about the D435.

closed 2018-05-05 by pirobot

aclswarm, 2019-09-25:
I tried this on a d435i running on a i7-6700 laptop and can confirm the impact of `png_level` within `depth/image_rect_raw/compressedDepth` dynamic reconfigure, with a 640x480 depth stream:

For `png_level` btwn 1-5, FPS is nearly constant, close to 30FPS.
At `png_level` 6, the FPS became sporadic btwn [25,31]
At `png_level` 7, the FPS became sporadic btwn [17,23]
At `png_level` 8, the FPS became sporadic btwn [7,10]
At `png_level` 9, the FPS became sporadic btwn [4,6]

Misticfury, 2020-11-03:
@aclswarm im having the same issue with latancy how do use the dynamic reconfigure to fix it i've never used it before?