realsenseai/realsense-ros #370: deploy ros realsense driver to apt-get

SteveMacenski, 2018-04-21:
It would make it much easier to use this software if there was a apt-get package for it so I dont have to build from source along with my code

RealSenseCustomerSupport, 2019-03-20:
----------------------------------------------

Close this issue for lack of activity. Please feel free to reopen if the issue is still relevant.

SteveMacenski, 2019-03-20:
This is still revelent, its waiting on actions from @RealSenseCustomerSupport

doronhi, 2019-03-21:
> This is still revelent, its waiting on actions from @RealSenseCustomerSupport

I agree.

RealSenseCustomerSupport, 2020-02-24:
----------------------------------------------

Closing this issue since there has been no activity. Please feel free to let us know  if the feature is still needed. Thank you.

SteveMacenski, 2020-02-24:
This feature is still needed and relevant. See above. Please stop closing issues without activity. Lack of activity is not a lack of issue, its a lack of interest in resolving it.

RealSenseCustomerSupport, 2020-02-24:
----------------------------------------------

No problem, we will keep this issue open for the apt-get package of the RealSense ROS Wrapper.
Thank you.

RealSenseCustomerSupport, 2020-05-05:
----------------------------------------------

Hi @SteveMacenski,
The latest wrapper 2.2.13 is available as ros-melodic-realsense2-camera and ros-kinetic-realsense2-camera.
https://github.com/IntelRealSense/realsense-ros#method-1-the-ros-distribution
Please give it a try. Thank you.

SteveMacenski, 2020-05-05:
there's now a new distribution: Noetic to support as well

RealSenseCustomerSupport, 2020-05-06:
----------------------------------------------

Hi @SteveMacenski, thank you for bring it to our attention about the upcoming (May 2020) ROS Noetic release.
The current realsense-ros wrapper still focus on supporting the LTS Melodic and Kinetic.

Please feel free to create a new feature request ticket for our development team to add the Noetic support. Thank you.

This case will be closed in 7 days unless otherwise. Thank you

SteveMacenski, 2020-05-06:
I think that still follows in the scope of this ticket

doronhi, 2020-05-06:
This issue's topic is "deploy ros realsense driver to apt-get" which we did so as far as I understand it, this issue can be closed now.
The request to support Noetic should be opened as a new issue. Naturally, now that we deployed the driver to apt-get, it can be assumed that support for Noetic will include that as well.
@SteveMacenski , It may seem like a technicality but it allows better tracking of the work being done.

closed 2020-05-06 by doronhi